rockowitz/ddcutil #436: DDC communication failed (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(15)])

constambeys, 2024-07-22:
ddcutil interrogate

Examining device /dev/i2c-4...
      Obtain and interpret EDID using normal i2c functions
      .....
      DDC communication succeeded

Do you know why interrogate works?

[interrogate.log](https://github.com/user-attachments/files/16337164/interrogate.log)

rockowitz, 2024-07-25:
Usually I suggest using options ***--ddcdata*** and ***-stats*** to see more details about the DDC data errors, but examining the journalctl section of the **interrogate** output I see the following line repeats: 
~~~
 Jul 22 18:30:32 dinos-pc-lan ddcutil[23299]: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
~~~

So the response packet to DDC requests is corrupt.  Why this is not the case with **interrogate** is unclear to me.

Your processor is relatively old (though not ancient).  That may contribute to the problem. 

If you have not already done so, try forcing a high sleep-multiplier using option ***--sleep-multiplier 2.0***. 

Finally, if you're comfortable building your own copy of the i915 driver, you could try making the changes described by 
Maciej Wójcik  in this   [i915 feature request](https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/11528).

dm17, 2025-01-30:
Same here
`DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA(10)])`

Lately it has been all of the time but previously I'd have many months in a row where it works. It used to be a reboot would get it working again. Is this display-specific or anything we can research about our display to determine more about how ddcutil interacts with it?