rockowitz/ddcutil #439: Cannot update beyond 1.2.2-1

samuelpietra, 2024-08-02:
Hey all,

Here's my tools:

- **OS:** Pop!_OS 22.04 LST x86_64
- **Kernel:** 6.9.3-76060903-generic
- **Shell:** zsh 5.8.1

I successfully installed `ddcutil` through `apt`, but could only change monitor brightness using `sudo`.

Reading the docs, I found that this was no longer needed in versions higher than **1.4.0**.

The odd thing is that apt doesn´t offer any other version than **1.2.2-1**.

I've also tried granting privileges to my user for the `i2c` group, but it only works until I end the terminal session.

Any thoughts?

rockowitz, 2024-08-02:
Your Pop release, 22.04, is based on Ubuntu 22.04 LST.  The version of ddcutil in that Ubuntu release is 22.04, which in turn reflects the version of ddcutil in Debian at the time Ubuntu 22.04 was built.

You can download more recent ddcutil packages from launchpad and openSuse.  See [Installing ddcutil and ddcui From Prebuilt Packages](https://www.ddcutil.com/install/). 

For configuring 1.2.2, see [Granting I2C Device Permissions on Versions Prior to 1.4.0](https://www.ddcutil.com/i2c_permissions/). Note that for release 1.2.2 the udev rule that should be in /etc/udev/rules.d or /usr/lib/udev/rules.d is 45-ddcutil-i2c.rules. 

Command **ddcutil environment** diagnoses problems with ddcutil configuration. 

If you are using command **usermod** to add your userid to group i2c and that does not persist across logins, it is a Pop issue. 

Alternatively, you can copy file 60-ddcutil-i2c.rules from a recent ddcutil branch and install it in directory /etc/udev/rules.d. That rule gives the currently logged on user RW access to /devi2c devices.

samuelpietra, 2024-08-13:
> Alternatively, you can copy file 60-ddcutil-i2c.rules from a recent ddcutil branch and install it in directory /etc/udev/rules.d. That rule gives the currently logged on user RW access to /devi2c devices.

That made the trick! I've just used the mentioned file from `2.1.4` release to my `/etc/udev/rules.d` directory, restarted the terminal session and it worked like a charm!

Thanks a lot!

closed 2024-08-13 by samuelpietra

PataviniMa, 2024-10-17:
@rockowitz I tried to install on Ubuntu 22.04 jammy via ppa, but I was getting an outdated version (1.3.0). Giving a look at the ppa packages, it seems that 2.1.4 was only released for noble. Is that intentional?