rockowitz/ddcutil #440: Samsung C24F390 - DDC communication failed

octvs, 2024-08-02:
Hey there, 

I can't get `ddcutil` to run on my monitor that is connected to a docking
station.

Setup:
- Samsung Monitor C24F390 
- ThinkPad Universal ThunderBolt 4 Dock

The monitor is connected in the following fashion: 

```txt
Monitor --(HDMI)--> KVM switch --(HDMI)--> Dock --(USB-C)--> Laptop
```

This might be related to #140, but also other issues related to docking
stations however I wasn't able to dig my way out of them. My apologies in case
I should have done dig better/deeper.

The result of the interrogate is attached.

[ddcutil-interrogate.txt](https://github.com/user-attachments/files/16472250/ddcutil-interrogate.txt)

rockowitz, 2024-08-02:
If it were me I'd try to isolate the problem by testing one connection at a time, e.g.: 
- laptop HDMI connector to monitor
- laptop USB-C connector to monitor (Thunderbolt type-C connectors support DisplayPort dual mode)
- laptop HDMI to KVM switch to monitor
- laptop HDMI to dock to monitor (no KVM)

octvs, 2025-11-12:
I no longer have this setup hence I can't help with debugging the issue. I think the best is to close this one.

closed 2025-11-12 by octvs