rockowitz/ddcutil #444: UGREEN Revodok Pro 308: ddcutil works with 1 display but not with 2

MichaelBrunn3r, 2024-08-29:
Hi, I'm connecting 2 external monitors via a Thunderbolt dock (UGREEN Revodok Pro 308) to my laptop and want to control their brightness. When I connect only 1 monitor via HDMI to the dock, I can control it's brightness via ddcutil:

```bash
> ddcutil detect --verbose
Display 1
   I2C bus:  /dev/i2c-13
      DRM connector:                         card1-DP-4
      /sys/class/drm/card1-DP-4/dpms         On
      /sys/class/drm/card1-DP-4/enabled      enabled
      /sys/class/drm/card1-DP-4/status       connected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true
      I2C address 0x37 (DDC)  responsive:    true
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-13/name       AUX USBC4/DDI TC4/PHY TC4
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-4/i2c-13
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                AG241QG4
      Product code:         9232  (0x2410)
      Serial number:
      Binary serial number: 91 (0x0000005b)
      Manufacture year:     2017,  Week: 10
      EDID version:         1.3
      Extra descriptor:
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.329
      Red   x,y:        0.644, 0.342
      Green x,y:        0.324, 0.626
      Blue  x,y:        0.153, 0.053
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 05 e3 10 24 5b 00 00 00   ...........$[...
      +0010   0a 1b 01 03 80 35 1e 78 2a e1 61 a4 57 53 a0 27   .....5.x*.a.WS.'
      +0020   0d 50 54 bf ef 00 d1 c0 81 80 31 68 31 7c 45 68   .PT.......1h1|Eh
      +0030   45 7c 61 68 61 7c 56 5e 00 a0 a0 a0 29 50 30 20   E|aha|V^....)P0
      +0040   35 00 0f 28 21 00 00 1e 40 e7 00 6a a0 a0 67 50   5..(!...@..j..gP
      +0050   08 20 98 04 0f 28 21 00 00 1a 00 00 00 fd 00 1e   . ...(!.........
      +0060   92 1e e6 3c 00 0a 20 20 20 20 20 20 00 00 00 fc   ...<..      ....
      +0070   00 41 47 32 34 31 51 47 34 0a 20 20 20 20 01 60   .AG241QG4.    .`
   VCP version:         2.2
   Controller mfg:      RealTek
   Firmware version:    0.1
   Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
   Current dynamic sleep adjustment multiplier:  1.30
```

But I cannot control either of the monitors if I connect the second one:

```bash
> ddcutil detect --verbose
Invalid display
   I2C bus:  /dev/i2c-13
      DRM connector:                         card1-DP-4
      /sys/class/drm/card1-DP-4/dpms         Off
      /sys/class/drm/card1-DP-4/enabled      disabled
      /sys/class/drm/card1-DP-4/status       disconnected
      Driver:                                i915
      I2C address 0x50 (EDID) responsive:    true
      I2C address 0x37 (DDC)  responsive:    true
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-13/name       AUX USBC4/DDI TC4/PHY TC4
      PCI device path:                       /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-4/i2c-13
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                AG241QG4
      Product code:         9232  (0x2410)
      Serial number:
      Binary serial number: 91 (0x0000005b)
      Manufacture year:     2017,  Week: 10
      EDID version:         1.3
      Extra descriptor:
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.329
      Red   x,y:        0.644, 0.342
      Green x,y:        0.324, 0.626
      Blue  x,y:        0.153, 0.053
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 05 e3 10 24 5b 00 00 00   ...........$[...
      +0010   0a 1b 01 03 80 35 1e 78 2a e1 61 a4 57 53 a0 27   .....5.x*.a.WS.'
      +0020   0d 50 54 bf ef 00 d1 c0 81 80 31 68 31 7c 45 68   .PT.......1h1|Eh
      +0030   45 7c 61 68 61 7c 56 5e 00 a0 a0 a0 29 50 30 20   E|aha|V^....)P0
      +0040   35 00 0f 28 21 00 00 1e 40 e7 00 6a a0 a0 67 50   5..(!...@..j..gP
      +0050   08 20 98 04 0f 28 21 00 00 1a 00 00 00 fd 00 1e   . ...(!.........
      +0060   92 1e e6 3c 00 0a 20 20 20 20 20 20 00 00 00 fc   ...<..      ....
      +0070   00 41 47 32 34 31 51 47 34 0a 20 20 20 20 01 60   .AG241QG4.    .`
   DDC communication failed
````

I tried to test all possible combinations, connecting each monitor individually, over HDMI and DP, but it always stops working if 2 monitors are connected.

From reading issues on this repo and a discussion you started on freedesktop, I think this should be DP MST issue? As far as I understand this is a driver issue and not something ddcutil can fix, but since this is a little over my head I wanted to ask for directions of where I can report this issue.

### Additional information

`ddcutil 2.1.4`
OS: `NixOS 24.05`
Graphics: `Intel Graphics (ADL GT2)` (i5-1240P)
DE: `GNOME 46` on Wayland
Kernel: `6.10.5`

rockowitz, 2024-08-29:
Yes, this looks like a MST issue.  With more information I may be able to say a bit more.  Please run **ddcutil interrogate** and submit the output as an attachment of some sort.  If it's convenient, please build from branch 2.1.5-dev, as it has improved diagnostics in this area.

MichaelBrunn3r, 2024-08-29:
Thx for taking the time. I'm unable to build 2.1.5-dev, because the instructions from the website fail:
```bash
> ./autogen.sh
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I config -I m4 --force -I m4 ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'config'.
libtoolize: copying file 'config/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal -I config -I m4 --force -I m4 ${ACLOCAL_FLAGS}
autoreconf: running: /nix/store/2jk7wsszi0p265vxbxmpvwnk5z04k7ii-autoconf-2.72/bin/autoconf --include=config --include=m4 --force
configure.ac:52: error: possibly undefined macro: AC_MSG_NOTICE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:226: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:640: error: possibly undefined macro: AC_MSG_WARN
autoreconf: error: /nix/store/2jk7wsszi0p265vxbxmpvwnk5z04k7ii-autoconf-2.72/bin/autoconf failed with exit status: 1
configure: error: cannot find required auxiliary files: config.guess config.sub compile ar-lib missing install-sh
```
Maybe I'm missing some dependencies?

But here is the output of 2.1.4:
[ddcutil-interrogate.txt](https://github.com/user-attachments/files/16808381/ddcutil-interrogate.txt)

rockowitz, 2024-08-31:
First, regarding the build problem, it appears that there's something unexpected about autotools configuration on nixos.  I presume that the 2.1.4 package you installed was somehow built on nixos, so I would look at how that build was done.

That said, the 2.1.4 interrogate output is sufficient for our purposes.

Yes, it's a driver issue. You're running a current kernel, which would have a recent i915 driver, so the problem probably lies in your particular hardware.  I suspect there's something about the UGREEN hub that's unexpected.  You can try posting to the i915 driver list.  They may have some diagnostics you can run, bit I'd not be too hopeful.

MichaelBrunn3r, 2024-08-31:
Thx for the pointers, I will try out posting on the i915 driver list.

In the meantime I borrowed a USB-C to HDMI adapter (also from UGREEN) and connected it to another thunderbolt port, but with that one DDC doesn't work at all. Maybe a UGREEN issue. I will look into finding a proven working adapter. Adding a 2nd cable isn't that bad of a backup solution.

Do you think the Linux Kernel DDCCI driver (https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux) will work any better? It was mentioned on some posts about DDC, but I can't test it rn, because its broken in NixOS :D

rockowitz, 2024-08-31:
I don't see how ddcci would make any difference.  Essentially it adds a layer on top of what driver i2c-dev (which ddcutil uses) does, The layer understands some DDC/CI features such as brightness, and presents the information through sysfs.  Multi-stream transport is handled deep down in the video driver layers of the call stack.

ZiemlichUndead, 2026-03-24:
Hey, I have the same issue with a UGREEN Revodok Pro 210.
So it seems likely it has something to do with UGREEN Docks.

I do however have a Laptop with AMD Vega 8 Graphics. So the pointer to i915 driver might not apply?
@MichaelBrunn3r did you get anything back from the i915 driver list?
I am happy to share any details, logs and such.

Btw I am using v2.2.6 and the only difference I see in my setup vs the issue is that when two monitors are connected, `ddcutil detect ` does not detect any of them. It just shows my laptops internal screen. And when i use `ddcutil detect --f21` it shows just one of the monitors with communication failure like in the issue.

[interrogate-1-monitor.txt](https://github.com/user-attachments/files/26228418/interrogate-1-monitor.txt)
[interrogate-2-monitors.txt](https://github.com/user-attachments/files/26228424/interrogate-2-monitors.txt)