rockowitz/ddcutil #445: Invalid checksum of Debian Trixie package

w8jcik, 2024-08-30:
`Dockerfile`

```
FROM debian:trixie

RUN apt update -y
RUN apt upgrade -y

RUN apt install curl gpg -y

RUN echo 'deb http://download.opensuse.org/repositories/home:/rockowitz/Debian_Testing/ /' | tee /etc/apt/sources.list.d/home:rockowitz.list
RUN curl -fsSL https://download.opensuse.org/repositories/home:rockowitz/Debian_Testing/Release.key | gpg --dearmor | tee /etc/apt/trusted.gpg.d/home_rockowitz.gpg > /dev/null

RUN apt update -y
RUN apt install ddcutil libddcutil-dev -y
```

```
docker build .
```

```
2.121 Error: Failed to fetch http://ftp.gwdg.de/pub/opensuse/repositories/home%3A/rockowitz/Debian_Testing/amd64/ddcutil_2.1.4-1_amd64.deb  Hash Sum mismatch
2.121    Hashes of expected file:
2.121     - SHA256:76c51aac52ad8ae50c7b626372a341f4d1f092368474261e1f9665cc4f6780ef
2.121     - SHA1:ea453f8fa6a66858e8bc0de859b3f9893168b55d [weak]
2.121     - MD5Sum:b4a5965317dbaf2ea03469b6ceda376e [weak]
2.121     - Filesize:397860 [weak]
2.121    Hashes of received file:
2.121     - SHA256:7f6e6bb0911146fd1a6df78d681d0d29312447bb0f3212519df9dd3736d136a0
2.121     - SHA1:dbe58f251b1d06abf6031975f265880d454f8649 [weak]
2.121     - MD5Sum:176e4a6fe4fd751069020cfecf111f10 [weak]
2.121     - Filesize:397860 [weak]
2.121    Last modification reported: Mon, 26 Aug 2024 03:46:52 +0000
2.121 Error: Failed to fetch http://ftp.gwdg.de/pub/opensuse/repositories/home%3A/rockowitz/Debian_Testing/amd64/libddcutil5_2.1.4-1_amd64.deb  Hash Sum mismatch
2.121    Hashes of expected file:
2.121     - SHA256:22ae46b270d3c7cc1c420a416b66cb559384a0b65bfd00b6736e7ff8d95cb1f7
2.121     - SHA1:063b59fe76ec607881589967a78aac32d14b0882 [weak]
2.121     - MD5Sum:3daf38d256b75f3ae47a0c62bc83386a [weak]
2.121     - Filesize:364772 [weak]
2.121    Hashes of received file:
2.121     - SHA256:0b458efa6676d818e6bebb1ac20f1693d6b8f98382fe78f77d1333d1bc96d57a
2.121     - SHA1:c80b514f52ffc7ab7efd3e0254e9ace481dce819 [weak]
2.121     - MD5Sum:0150fddcb0288008139d8ec249709064 [weak]
2.121     - Filesize:364772 [weak]
```

rockowitz, 2024-08-31:
Not sure what to do with this.  The hashes are managed automatically by OBS.

w8jcik, 2024-09-01:
I am not sure if the issue was fixed or fixed itself, but the checksum no longer fails.

Thank you for help.

closed 2024-09-01 by w8jcik