rockowitz/ddcutil #446: DDC Communication failed. [EIO in ddc_write_read_with_retry, causes: EIO]

checkinindza, 2024-08-31:
Hi, so I'm trying to control my external display's brightness with ddcutil, but the issue is, it says `Display not found`, running `sudo ddcutil detect` shows `DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])`

The display is Lenovo T24i-10 connected through a Dell WD15 dock's HDMI port. 

`sudo ddcutil interrogate --verbose ` output

[output.log](https://github.com/user-attachments/files/16826064/output.log)

If it changes anything, I'm using proprietary NVIDIA drivers.

rockowitz, 2024-09-01:
The i2detect output in the interrogate log shows that I2C slave address x50 (EDID) is responsive on /dev/i2c-2, but slave address x37 (DDC) is not.  (In case you're unfamiliar with it, i2cdetect is part of the i2c-tools package, so this test is independent of ddcutil.)

Possibility 1: Your monitor does not support DDC/CI. Unlikely, since the spec sheet for the monitor indicates DDC support, so I mention this for completeness.   Does DDC/CI work if the monitor is plugged directly into the iGPU or another computer without the dock?  Some monitors have a setting in the OSD that enables/disables DDC/CI.

Possibility 2:  The dock does not support DDC/CI.  Can you check it with another system without the Nvidia driver?

Possibility 3: The problem is in the Nvidia driver's Multi-Stream-Transport implementation.  I would not be surprised if this is the case, but I have not tested this configuration. Perhaps someone who sees this issue and who uses the proprietary Nvidia driver can speak to whether DDC works with MST.

checkinindza, 2024-09-02:
1. Yeah, I did check the monitor's spec sheet before posting this to make sure myself. I did find on the user guide that you can actually just click a button on the monitor for a few seconds and it should disable/enable DDC/CI. And I did that to make sure DDC/CI is enabled. 

2. Well, I do not have another computer with Linux installed. But I do have a work issued all Intel laptop. I tried to use `ddcutil` on EndeavourOS LiveUSB, but `ddcutil capabilities` showed `Display not found`. Monitor was connected directly to the laptop. I'll attach a log, if it helps, but I'm guessing there are just things missing on it, but it did have `ddcutil` and `i2c-tools` already installed. And to test if DDC/CI works through the dock, I don't know how accurate this experiment is, but I used a Windows tray app called Twinkle Tray (https://twinkletray.com/) on the same laptop with monitor connected to the dock. If I understand correctly Twinkle Tray uses DDC/CI to do that and it worked fine. I used Twinkle Tray both on my own laptop (it's a dual-boot setup) and the work issued one. Worked fine on both.

Sadly, my own laptop's display outputs are all wired directly to the NVIDIA GPU, so I can't really test on it connecting the monitor to the iGPU.

[output.log](https://github.com/user-attachments/files/16837649/output.log)