rockowitz/ddcutil #448: ddcutil detect is unreliable in detecting monitors

SmurgBurglar, 2024-09-14:
ddcutil often fails to detect my monitor's LG ULTRAGEAR+ and PHL 243V5, i have uploaded the output of ddcutil interrogate.
[ddcutil_interrogate.log](https://github.com/user-attachments/files/17002010/ddcutil_interrogate.log)

rockowitz, 2024-09-16:
First thing to check is that failures are not happening because dynamic sleep adjustment is being too aggressive. Try using option ***--disable-dynamic-sleep***, which will cause ddcutil to use the sleep timings in the DDC/CI spec.  If that doesn't work, try ***--sleep-multiplier 2.0**, which will cause ddcutil to use double the specified sleep timings.  

Option ***---stats*** provides aggregate reports on the errors encountered.  In the case of DDC data errors (i.e. corrupt data in a DDC/CI reply packet), option ***--ddcdata*** will report the error.

dm17, 2025-01-30:
@rockowitz what command do we run `--disable-dynamic-sleep` with?

Here is the output of the command I'm trying to run that only sometimes works (and I'm not sure why) plus `--stats`
```
ddcutil setvcp 10 - 45 --stats
Display not found

EXECUTION STATISTICS

Maxtries Settings:
Operation Type                    Current  Default     Min     Max
Write only exchange tries:              4        4       4       4
Write read exchange tries:             10        4      10      10
Multi-part read exchange tries:         8        4       8       8
Multi-part_write exchange tries:        8        4       8       8

Retry statistics for write only
   No tries attempted

Retry statistics for write-read
   Max tries allowed: 10
   Successful attempts by number of tries required: None
   Total successful attempts:          0
   Failed due to max tries exceeded:   1
   Failed due to fatal error:          0
   Total attempts:                     1

Retry statistics for multi-part read
   No tries attempted

Retry statistics for multi-part write
   No tries attempted

DDC Related Errors:

Count   Status Code                          Description
   10   DDCRC_DDC_DATA               (-3001) DDC data error
    1   DDCRC_RETRIES                (-3007) maximum retries exceeded  (derived)
Total errors: 11

Errors Wrapped in Retry:

Count   Status Code                          Description
   10   DDCRC_DDC_DATA               (-3001) DDC data error
Total errors: 10

IO and Sleep Events:
   Total IO events:        121
   IO error count:          11
   Total sleep events:      20

   Sleep Event type      Count
   SE_WRITE_TO_READ             10
   SE_POST_WRITE                 0
   SE_POST_READ                 10
   SE_POST_SAVE_SETTINGS         0
   SE_PRE_MULTI_PART_READ        0
   SE_POST_CAP_TABLE_SEGMENT     0
   SE_SPECIAL                    0

Call Stats:
   Type                                     Count    Millisec  (      Nanosec)
   i2c writes using ioctl (I2_IOCTL_WRITE)     71         264  (    264972769)
   i2c reads using ioctl  (I2_IOCTL_READ)      22         170  (    170059048)
   open file calls        (IE_OPEN)            14           0  (        28855)
   close file calls       (IE_CLOSE)           14           0  (        13264)
   Totals:                                    121         435  (    435073936)

Sleep Call Stats:
   Total sleep calls:                                      20
   Requested sleep time milliseconds :                   2000
   Actual sleep milliseconds (nanosec):                  2006  (   2006854663)

Total elapsed milliseconds (nanoseconds):                2497  (   2497613080)

Total non sleep system call time:        435 milliseconds
Total sleep call time:                  2006 milliseconds
Elapsed time:                           2497 milliseconds
```