rockowitz/ddcutil #450: Can't set brightness on AOC Monitor

AnuraagReddy123, 2024-09-21:
I am trying to set the brightness for my monitor using
```
sudo ddcutil -d 1 setvcp 10 60 -v
```
I get the following output
```
Feature definition file not found: AOC-22B2HM2B-8706.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
```
However the following command works as intended.
```
$ sudo ddcutil -d 1 getvcp 10
VCP code 0x10 (Brightness                    ): current value =    20, max value =   100
```
This is the output of `sudo ddutil detect`
```
Display 1
   I2C bus:  /dev/i2c-2
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               AOC - UNK
      Model:                22B2HM2B
      Product code:         8706  (0x2202)
      Serial number:        ZZ3Q4HA002303
      Binary serial number: 2303 (0x000008ff)
      Manufacture year:     2024,  Week: 14
   VCP version:         2.1

Invalid display
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               AUO - UNK
      Model:
      Product code:         20107  (0x4e8b)
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2020,  Week: 0
   This is a laptop display.  Laptop displays do not support DDC/CI
```
System Info
- Interrogate: [interrogate.txt](https://github.com/user-attachments/files/17082995/interrogate.txt)
- Environment: [environment.txt](https://github.com/user-attachments/files/17083001/environment.txt)
- Capabilities: [capabilities.txt](https://github.com/user-attachments/files/17082996/capabilities.txt)

rockowitz, 2024-09-25:
It appears that setvcp is properly setting the value, but the verification sub-operation is failing.   Try running  **ddcutil  setvcp 10 60 --verbose  --trcftom ddc_set_verified_vcp_value_with_retry  --stats** and submit the output as an attachment.  Thank you.

AnuraagReddy123, 2024-09-25:
I am getting this output
```
$ sudo ddcutil setvcp 10 60 --verbose --trcfrom ddc_set_verified_vcp_value_with_retry --stats
Traced call stack function not found: ddc_set_verified_vcp_value_with_retry
```

rockowitz, 2024-09-25:
I didn't read your report closely enough. Your setvcp command should have set the value to 60.  But the getvcp command reports 10.  So it would appear that the message that verification failed is correct.

Might the monitor be in a mode (e.g. cinema) that doesn't allow you to change the brightness?

AnuraagReddy123, 2024-09-25:
That's it! It was set to Text Reading mode that had a fixed brightness. Setting it back to standard worked and then running the setvcp command worked

closed 2024-09-28 by rockowitz

ratijas, 2024-10-21:
It took me a while to realize this while I was rewriting [DDCUtil QML frontend](https://github.com/d-vogel/QMLddcutil), but apparently sRGB mode on AOC monitor is hardcoded to a certain backlight brightness, so you can't really change it. (Now imagine me debugging a seemingly "disfunctional" GUI slider for five hours straight :skull: )