rockowitz/ddcutil #451: 1.4.1 not working all the time

foilandwater, 2024-09-21:
My setup is a pi 4 running bookwork and 1.4.1, installed using apt-get.

I have a Samsung monitor connected via HDMI and I am able to issue commands that control brightness and turn the display on and off - the brightness commands always work, but the power on/off don't always work. For example: 

ddcutil setvcp D6 01 will either work, or give an error like:

Setting value failed for feature d6, rc=DDCRC_RETRIES(-3007): maximum retries exceeded
    Try errors: DDCRC_RETRIES

or

Setting value failed for feature d6, rc=DDCRC_NULL_RESPONSE(-3002): received DDC null response

Sometimes when it gives the error, it will still perform the action of turning the display on or off. Sometimes not.

ddcutil getvcp D6 works and always returns the current status of the display.

I have tried adding --display 1 as well as targeting the exact model/serial number directly, but it still has the same erratic behaviour of either working or not working.
"ddcutil --model="S27C31x" --sn="H9DW900554" setvcp d6 1"

Usually if I run the command to turn on or off 3 or 4 times in a row, it eventually works.

This is an amazing app, please let me know if anyone has any suggestions for how to get this working reliably.

Thanks,
Sean.

rockowitz, 2024-09-21:
Looks like your I2C communication is marginal.  Using option ***--stats*** and ***--ddc*** should give a better sense of where the errors are happening.   Try explicitly setting a high sleep multiplier, e.g. ***--sleep-multiplier 2.0***.

foilandwater, 2024-09-22:
Thank you for the reply and the info. I found another post where someone wrote a for loop which makes it a reliable solution for me. Thank you again for this amazing app.

# power off
for i in {1..99} ; do
    ddcutil setvcp d6 x04 && break
done

# power on
for i in {1..99} ; do
    ddcutil setvcp d6 x01 && break
done

closed 2024-09-22 by foilandwater