rockowitz/ddcutil #452: Setting sleep multiplier crashes ddcutil

pacjo, 2024-09-22:
Hi, I once again connected my old samsung monitor (CF390) which has issues with ddc/ci. There were suggestions to set sleep multiplier and I remember doing that before, but when trying to do that I run into an issue.

### logs

nushell:
```console
~ ❯ ddcutil detect --sleep-multiplier=5
ddcutil: dsa2.c:930: dsa2_record_final: Assertion `rtable->cur_retry_loop_step <= step_last' failed.
Error: nu::shell::coredump_error

  × program coredump error
   ╭─[entry #37:1:1]
 1 │ ddcutil detect --sleep-multiplier=5
   · ───┬───
   ·    ╰── coredump detected. received signal: 6
   ╰────

~ ❯ 

```

bash:
```console
~ ❯ ddcutil detect --sleep-multiplier=5
ddcutil: dsa2.c:931: dsa2_record_final: Assertion `rtable->cur_step <= rtable->cur_retry_loop_step' failed.
Aborted (core dumped)
```

Same happens with sudo. Running on nixos unstable

```
ddcutil 2.1.4
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

rockowitz, 2024-09-22:
I believe this is the same problem as issue #407.  It was fixed by commit b4c1a48e36571250cab42b39d86201e7b654dfcd in branch 2.1.5-dev.

pacjo, 2024-09-22:
Thanks, I didn't see that.

Replacing `ddcutil` with
```nix
    (pkgs.ddcutil.overrideAttrs (oldAttrs: rec {
      src = pkgs.fetchFromGitHub {
        owner = "rockowitz";
        repo = "ddcutil";
        rev = "2.1.5-dev";
        sha256 = "sha256-oLnfeyg/UWvxahH+uu8tk7WB8xdyjoxZs+/NfsiKNSk=";    # latest at the time of writing
      };
    }))
```
in my configuration does fix the issue. Closing.

closed 2024-09-22 by pacjo