rockowitz/ddcutil #454: Unable to get capabilities for LG Ultrafine 22MD4KA - VCP detection failed

Merlin04, 2024-09-25:
I'm encountering an error when trying to detect the capabilities of my LG Ultrafine 22MD4KA monitor. When I run `sudo ddcutil detect` I get:

```
[...]
Display 2
   I2C bus:  /dev/i2c-15
   DRM connector:           card1-DP-2
   EDID synopsis:
      Mfg id:               GSM - Goldstar Company Ltd (LG)
      Model:                LG UltraFine
      Product code:         23312  (0x5b10)
      Serial number:        801NTEP9M516
      Binary serial number: 327516 (0x0004ff5c)
      Manufacture year:     2018,  Week: 1
   VCP version:         Detection failed
```

and when running `sudo ddcutil -d 2 -v capabilities` I get:

```
/dev/i2c-15, Feature 0xdd should not exist, but the monitor reports it as valid
Feature definition file not found: GSM-LG_UltraFine-23312.mccs
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.
Unable to get capabilities for monitor on Display_Handle[i2c-15: fd=3].  Maximum DDC retries exceeded.
```

Using ddcutil to control the brightness of the monitor works fine, but prints the VCP error again:

```
> sudo ddcutil -d 2 setvcp 10 45
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.
```

Here's my [ddcutil-interrogate.log](https://github.com/user-attachments/files/17137329/ddcutil-interrogate.log).

rockowitz, 2024-09-28:
Do you know if the errors occurred for ddcutil 1.4.5 or earlier?

Does using option ***--disable-dynamic-sleep*** resolve the problems?

What is the output from **ddcutil capabilities --display 2 --trcfunc get_capabilities_info_buffer --excp**

What is the output from **ddcutil --display 2 getvcp df  --verbose --excp**

rockowitz, 2024-11-18:
User @adlr encountered a similar situation with LG Ultrafine monitors in issue #468.  The monitor reports that feature xDF (Vcp Version) is unsupported.  So the message: 
~~~
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.
~~~
is correct.  However, it does not provide information relevant to command **setvcp**.  It should only be reported only in the output of commands such as  **getvcp** that need to interpret feature values.

SinTrans, 2025-01-05:
I'm also encountering this error. `--disable-dynamic-sleep` didn't seem to resolve the problem.

```
$ sudo ddcutil --disable-dynamic-sleep detect   

[...]
Display 1                                                                                                            
   I2C bus:  /dev/i2c-14                                                                                             
   DRM connector:           card1-DP-1                                                                               
   EDID synopsis:                                                                                                    
      Mfg id:               GSM - Goldstar Company Ltd (LG)                                                          
      Model:                LG UltraFine                                                                             
      Product code:         23419  (0x5b7b)                                                                          
      Serial number:        201NTUWGL126                                                                             
      Binary serial number: 564126 (0x00089b9e)                                                                      
      Manufacture year:     2022,  Week: 1                                                                           
   VCP version:         Detection failed       
```

```
$ sudo ddcutil capabilities --display 1 --trcfunc get_capabilities_info_buffer --excp                                
Traced function not found: get_capabilities_info_buffer          
```

```
$ sudo ddcutil --display 1 getvcp df --verbose --excp
Feature definition file not found: GSM-LG_UltraFine-23419.mccs
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.

Getting data for non-table VCP code 0xdf - VCP Version:
VCP code 0xdf (VCP Version                   ): Unsupported feature code
(ddc_get_formatted_value_for_dfm) Freeing exception:
   Exception in function ddc_get_nontable_vcp_value: status=DDCRC_REPORTED_UNSUPPORTED(-3005): DDC reports facility unsupported
      Unsupported feature
```