rockowitz/ddcutil #457: Key Expired Opensuse repo for Ubuntu 22.04

PaulWentzel, 2024-10-07:
Maybe i am just too dumb but when i try to update it tells me that the signing key has expired. When trying to readd the key, according to https://software.opensuse.org/download.html?project=home%3Arockowitz&package=ddcutil  it doesnt help. Pls help.

rockowitz, 2024-10-13:
There are multiple problems with OBS builds for Debian based systems such as Ubuntu.  These do not appear to be ones I can fix.  I've added a message to the web page on using prebuilt packages.

You should be able to download prebuilt packages from the launchpad PPA.  Builds there are marked as being associated with a particular Ubuntu release, but in fact the most recent build should work for any Ubuntu or Debian version.

PataviniMa, 2024-10-18:
@rockowitz I'm on 22.04 jammy, too. If I try to force the ppa distribution to noble (in order to get the most recent version of ddcutil), then I get dependency issues:

> Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of Incoming.
> The following information may help to resolve the situation:

> The following packages have unmet dependencies:
>  ddcutil : Depends: libc6 (>= 2.38) but 2.35-0ubuntu3.8 is to be installed
>            Depends: libglib2.0-0t64 (>= 2.75.3) but it is not installable
>            Depends: libjansson4 (>= 2.14) but 2.13.1-1.1build3 is to be installed

Any ideas?

rockowitz, 2024-10-20:
@PataviniMa  I have added builds of ddcutil 2.1.4 for 22.04 jammy (they have built successfully but at this moment they have not yet been published).  Let me know if this works for you. 

I have also added builds for 24.10 noble and 25.04 plucky (beta).

PataviniMa, 2024-10-20:
@rockowitz Worked like a charm, thank you so much!