rockowitz/ddcutil #465: Thinkpad X13 Gen 4 (AMD) + Dell S2722DC not detected in 2.1.5

YaroST12, 2024-11-03:
Running on Manjaro Linux. Installed 2.1.4 - the monitor is detected fine but I had an issue after a reboot, ddcutil seem to have locked up the system after logging in. Decided to dig in, cloned 2.1.5-dev - the display connected over USB-C is not detected anymore.

The output of interrogate for 2.1.4 and 2.1.5:
[ddcutil_2_1_4_dev_integgorate.txt](https://github.com/user-attachments/files/17612533/ddcutil_2_1_4_dev_integgorate.txt)
[ddcutil_2_1_5_dev_integgorate_d43bbd8bf9ff2a57b240f1cb03620b44578dd665.txt](https://github.com/user-attachments/files/17612534/ddcutil_2_1_5_dev_integgorate_d43bbd8bf9ff2a57b240f1cb03620b44578dd665.txt)

I will check if I can reproduce the freeze on 2.1.4 and open a separate issue for it. Tried bisecting but mid-feature hashes are barely working.

rockowitz, 2024-11-04:
The display detection code has seen a lot of rework lately.  Please be sure your copy of branch 2.1.5-dev is current with that on github.  

One thing that **interrogate** does is to scan the journal for lines having a word of interest.   I can see parts of **ddcutil** backtraces.  Please submit the full journal for the current boot when **ddcutil** fails to detect the Dell monitor. If you'd prefer, find the lines for the **ddcutil** backtraces and just send  a segment starting several lines before the backtraces and ending several lines after.

Try option: ***--disable-try-get-edid-from-sysfs***. This disables some of the new logic in display detection.  Does this make a difference?

Try option: ***--use-file-io*** which uses a different interface to kernel module i2c-dev. 

Execute **ddcutil** with options **--trcfunc i2c_check_bus2 --trcfunc check_x37_for_businfo --trcfunc is_laptop_for_businfo --trcfile i2c_execute*** and submit the output.

rockowitz, 2024-11-04:
Conceivably this [bug](https://gitlab.freedesktop.org/drm/amd/-/issues/3672) that I reported to the amdgpu list could be related to the lockup you've seen.  If the system locks up can you ssh into it?  If so, what's in dmesg and the journal?

YaroST12, 2024-11-04:
I've just compiled ddcutil from commit d5ed0506fa38f919c807145678dfd61b8486b9c8, still no detection.

Output of `ddcutil interrogate --brief --disable-try-get-edid-from-sysfs --use-file-io` from this version:
[ddcutil_2_1_5_dev_integgorate_d5ed0506fa38f919c807145678dfd61b8486b9c8.txt](https://github.com/user-attachments/files/17616776/ddcutil_2_1_5_dev_integgorate_d5ed0506fa38f919c807145678dfd61b8486b9c8.txt)

Output of `ddcutil detect --brief --trcfunc i2c_check_bus2 --trcfunc check_x37_for_businfo --trcfunc is_laptop_for_businfo --trcfile 'i2c_execute*'`:
[ddcutil_2_1_5_dev_integgorate_trcfile_d5ed0506fa38f919c807145678dfd61b8486b9c8.txt](https://github.com/user-attachments/files/17616821/ddcutil_2_1_5_dev_integgorate_trcfile_d5ed0506fa38f919c807145678dfd61b8486b9c8.txt)

I don't have anything to ssh into the system from, I can't switch to a tty either. I can no longer reproduce it even on 2.1.4 version, will see how it goes. The only ddcutil related messages that I can see in the log:

`Nov 04 01:14:41 yaro systemd-coredump[112283]: Process 112282 (ddcutil) of user 1000 terminated abnormally with signal 6/ABRT, processing...
Nov 04 01:14:41 yaro systemd-coredump[112284]: [🡕] Process 112282 (ddcutil) of user 1000 dumped core.
                                               
                                               Stack trace of thread 112282:
                                               #0  0x0000558391b153f4 n/a (libc.so.6 + 0x963f4)
                                               #1  0x0000558391abc120 raise (libc.so.6 + 0x3d120)
                                               #2  0x0000558391aa34c3 abort (libc.so.6 + 0x244c3)
                                               #3  0x0000558391aa33df n/a (libc.so.6 + 0x243df)
                                               #4  0x0000558391ab4177 __assert_fail (libc.so.6 + 0x35177)
                                               #5  0x0000558392069719 i2c_check_bus2 (ddcutil + 0x76719)
                                               #6  0x0000558392069826 i2c_non_async_scan (ddcutil + 0x76826)
                                               #7  0x00005583920699b4 i2c_detect_buses0 (ddcutil + 0x769b4)
                                               #8  0x0000558392069b25 i2c_detect_buses (ddcutil + 0x76b25)
                                               #9  0x000055839204c970 ddc_detect_all_displays (ddcutil + 0x59970)
                                               #10 0x000055839204d1a4 ddc_ensure_displays_detected (ddcutil + 0x5a1a4)
                                               #11 0x000055839203579b main (ddcutil + 0x4279b)
                                               #12 0x0000558391aa4e08 n/a (libc.so.6 + 0x25e08)
                                               #13 0x0000558391aa4ecc __libc_start_main (libc.so.6 + 0x25ecc)
                                               #14 0x0000558392035a65 _start (ddcutil + 0x42a65)
                                               ELF object binary architecture: AMD x86-64
`

YaroST12, 2024-11-04:
Something interesting in the journal:

```
Nov 04 12:12:33 yaro systemd[1]: fprintd.service: Deactivated successfully.
Nov 04 12:12:19 yaro ddcutil[4410]: dh=Display_Handle[i2c-14: fd=3], Replacing adjusted sleep multiplier 0.70 with 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
Nov 04 12:12:18 yaro ddcutil[4410]: busno=14, All features that should not exist detected. Monitor does not indicate unsupported
Nov 04 12:12:45 yaro telegram-desktop[3399]: qt.gui.imageio.jpeg: Corrupt JPEG data: premature end of data segment
Nov 04 12:12:36 yaro systemd-timesyncd[1047]: Contacted time server 178.239.19.61:123 (3.manjaro.pool.ntp.org).
Nov 04 12:12:33 yaro systemd[1]: fprintd.service: Deactivated successfully.
Nov 04 12:12:19 yaro ddcutil[4410]: dh=Display_Handle[i2c-14: fd=3], Replacing adjusted sleep multiplier 0.70 with 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
Nov 04 12:12:18 yaro ddcutil[4410]: busno=14, All features that should not exist detected. Monitor does not indicate unsupported
Nov 04 12:12:18 yaro ddcutil[4410]: busno=14, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Nov 04 12:12:18 yaro gnome-shell[4418]: No monitor detected on bus /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]: (i2c_open_bus                  )           Cross instance locking failed
Nov 04 12:12:18 yaro gnome-shell[4418]:    sh      4444 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    ddcutil 4418 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    ddcutil 4410 yaro 3uW  CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    COMMAND  PID USER FD   TYPE DEVICE SIZE/OFF NODE NAME
Nov 04 12:12:18 yaro ddcutil[4418]: Processes locking /dev/i2c-14 (inode 540):
Nov 04 12:12:18 yaro ddcutil[4418]:    sh      4453 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro ddcutil[4418]:    ddcutil 4418 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro ddcutil[4418]:    ddcutil 4410 yaro 3uW  CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro ddcutil[4418]:    COMMAND  PID USER FD   TYPE DEVICE SIZE/OFF NODE NAME
Nov 04 12:12:18 yaro gnome-shell[4418]: Processes locking /dev/i2c-14 (inode 540):
Nov 04 12:12:18 yaro gnome-shell[4418]:    sh      4453 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    ddcutil 4418 yaro 3u   CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    ddcutil 4410 yaro 3uW  CHR  89,14      0t0  540 /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]:    COMMAND  PID USER FD   TYPE DEVICE SIZE/OFF NODE NAME
Nov 04 12:12:18 yaro ddcutil[4410]: busno=14, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Nov 04 12:12:18 yaro ddcutil[4410]: busno=14, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Nov 04 12:12:18 yaro ddcutil[4418]: Programs holding /dev/i2c-14 open:
Nov 04 12:12:18 yaro ddcutil[4418]: Max wait exceeded for /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]: Programs holding /dev/i2c-14 open:
Nov 04 12:12:18 yaro gnome-shell[4418]: Max wait exceeded for /dev/i2c-14
Nov 04 12:12:18 yaro gnome-shell[4418]: (i2c_open_bus                  )           busno=14, flock() returned: EAGAIN(-11): Resource temporarily unavailable
Nov 04 12:12:18 yaro nordvpnd[1080]: 2024/11/04 12:12:18 [NC] Connected
Nov 04 12:12:18 yaro gnome-shell[2247]: See https://wayland.freedesktop.org/libinput/doc/1.26.2/touchpad-jumping-cursors.html for details
Nov 04 12:12:18 yaro gnome-shell[2247]: libinput error: event12 - SYNA88022:00 06CB:CE67 Touchpad: kernel bug: Touch jump detected and discarded.
Nov 04 12:12:18 yaro nordvpnd[1080]: 2024/11/04 12:12:18 [Info] [NC] Connected
Nov 04 12:12:18 yaro ddcutil[4402]: dh=Display_Handle[i2c-14: fd=3], Replacing adjusted sleep multiplier 0.70 with 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
Nov 04 12:12:18 yaro ddcutil[4402]: busno=14, All features that should not exist detected. Monitor does not indicate unsupported
Nov 04 12:12:18 yaro ddcutil[4402]: busno=14, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Nov 04 12:12:18 yaro gnome-shell[4418]: (i2c_open_bus                  )           busno=14, flock() returned: EAGAIN(-11): Resource temporarily unavailable
```
Admittedly, this is from a successful boot attempt, but looks a bit suspicious.

rockowitz, 2024-11-06:
There's a lot to unpack here.  First, let me apologize for an errant asterisk.  I meant to type --trcfile i2c_execute, not i2c_execute* in the command options.  The extra asterisk came from an error in markdown formatting.

I've added additional tracing to function i2c_check_bus2(), so please build from the latest 2.1.5-dev branch.

Try executing command **ddcutil detect --verbose --trcfrom i2c_check_bus2 ** and submit the output.  The output will be voluminous, but should help me locate the problem. Note that I'm no longer interested in the output from **interrogate**.

The coredump shows that two copies of ddcutil are executing and are conflicting.  Do you know what the other copy is? If you add option ***--disable-flock*** does that make a difference?   And also, does using options ***--disable-try-get-edid-from-sysfs --f21*** make a difference?

YaroST12, 2024-11-07:
One copy is installed from aur, the other one is that I built. Bootup issues seem to stem from the amdgpu bug that you opened a while ago, I see the same dcn31 related warnings.

To your suggestions:

```
ddcutil detect --brief --disable-try-get-edid-from-sysfs --f21
No displays found.

```
And my laptop's display froze up until the command finished.

--disable-flock - option not found.

Output from the "ddcutil detect --verbose --trcfrom i2c_check_bus2" command: [ddcutil_2_1_5_dev_detect_690d9bd1122b40b2615e811736db6d0daab1e732.txt](https://github.com/user-attachments/files/17670030/ddcutil_2_1_5_dev_detect_690d9bd1122b40b2615e811736db6d0daab1e732.txt)

rockowitz, 2024-11-18:
My oops.  The proper option is ***--disable-cross-instance-locks***.

rockowitz, 2024-11-23:
I have replaced, in branch 2.1.5-dev,  the function that finds the display adapter for an i2c device.  Does **ddcutil detect --verbose** find the display?  If not, run **ddcutil detect --verbose --trcfunc sysfs_find_adapter" and submit the output. 

This should fix display detection, but does not address the other problems you described.