rockowitz/ddcutil #467: DDC communication fails with a Lenovo T24i-10

joshuajeschek, 2024-11-06:
I have a [Lenovo T24i-10](https://support.lenovo.com/us/en/solutions/pd500248-thinkvision-t24i-10-61ce-monitor-overview) connected via HDMI and get an error when using `ddcutil detect`. I think the monitor should support DDC, but right now, it doesn't work with `ddcutil` On my Samsung C34J79x `ddcutil` works as expected.

Output for `ddcutil detect`:
```
Invalid display
   I2C bus:  /dev/i2c-7
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               LEN - Lenovo Group Limited
      Model:                T24i-10
      Product code:         24998  (0x61a6)
      Serial number:        VT518162
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 22
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(10)])

Display 1
   I2C bus:  /dev/i2c-10
   DRM connector:           card1-DP-2
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                C34J79x
      Product code:         3869  (0x0f1d)
      Serial number:        H4ZN501861
      Binary serial number: 1129796436 (0x43575354)
      Manufacture year:     2038,  Week: 50
   VCP version:         2.1
```

Output of `i2cdetect -y 7`, maybe it is of interest:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

Output of `ddcutil interrogate`: [interrogate.txt](https://github.com/user-attachments/files/17645335/interrogate.txt)

Please let me know if I can supply you with more information.

rockowitz, 2024-11-18:
Attempting to read feature X10 (Brightness) is part of the tests as to whether the monitor supports DDC;  I've yet to see a monitor that does not implement that feature.  Strangely, it is however readable when performed by command **probe**.  What is the output of **ddcutil --bus 7 getvcp 10 --verbose**?

rockowitz, 2024-11-18:
Also, as a workaround, consider using option ***--skip-ddc-checks***, introduced in release 2.1.0.  Testing DDC communication occurs in two parts.  At the basic I2C layer, there's a simple check that I2C slave address x37 (DDC) is alive.  At the higher DDC layer, communication is tested by actually attempting to read feature x10 (which should always exist) and also to read several features that are expected not to exists, to check for the possibility that the monitor reports unsupported features correctly.  Option ***--skip-ddc-checks*** skips the latter tests.  (The option was introduced to improve performance of scripts invoking **ddcutil** repeatedly for the same device.)

joshuajeschek, 2024-11-20:
> Attempting to read feature X10 (Brightness) is part of the tests as to whether the monitor supports DDC; I've yet to see a monitor that does not implement that feature. Strangely, it is however [...]

```
Feature definition file not found: LEN-T24i_10-24998.mccs
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.

Getting data for non-table VCP code 0x10 - Brightness:
VCP code 0x10 (Brightness                    ): Maximum retries exceeded
```
but sometimes it also only prints this:
```
DDC communication failed for monitor on bus /dev/i2c-7
```

joshuajeschek, 2024-11-20:
Using `--skip-ddc-checks` results in the following output of `ddcutil detect`:
```
Display 1
   I2C bus:  /dev/i2c-7
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               LEN - Lenovo Group Limited
      Model:                T24i-10
      Product code:         24998  (0x61a6)
      Serial number:        VT518162
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 22
   VCP version:         Detection failed

Display 2
   I2C bus:  /dev/i2c-10
   DRM connector:           card1-DP-2
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                C34J79x
      Product code:         3869  (0x0f1d)
      Serial number:        H4ZN501861
      Binary serial number: 1129796436 (0x43575354)
      Manufacture year:     2038,  Week: 50
   VCP version:         2.1
```

However, observer the outputs of these commands:
```
> ddcutil -l T24i-10 getvcp 10
Display not found

> ddcutil -l T24i-10 getvcp 10 --skip-ddc-checks
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.
VCP code 0x10 (Brightness                    ): Maximum retries exceeded
```
The error is a different one, but it still fails to set the brightness.

rockowitz, 2024-11-22:
All of these errors reflect DDC communication failure.  Please submit the output of **ddcutil detect --verbose --ddcdata**.  This should show the bytes of the packets returned by the monitor.

joshuajeschek, 2024-11-23:
> All of these errors reflect DDC communication failure. Please submit the output of **ddcutil detect --verbose --ddcdata**. This should show the bytes of the packets returned by the monitor.

```
DDC: Unexpected source address 0x80, should be 0x6e
DDC: i2c_response_bytes: 80 80 be ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: Unexpected VCP opcode 0x10, should be 0xdd, response data bytes: 02001000 00640064
Invalid display
   I2C bus:  /dev/i2c-7
      DRM connector:                         card1-HDMI-A-1
      /sys/class/drm/card1-HDMI-A-1/dpms     On
      /sys/class/drm/card1-HDMI-A-1/enabled  enabled
      /sys/class/drm/card1-HDMI-A-1/status   connected
      Driver:                                amdgpu
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-7/name        AMDGPU DM i2c hw bus 2
      PCI device path:                       /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/0000:02:00.0/0000:03:00.0/i2c-7
   EDID synopsis:
      Mfg id:               LEN - Lenovo Group Limited
      Model:                T24i-10
      Product code:         24998  (0x61a6)
      Serial number:        VT518162
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 22
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0xa0 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.306, 0.329
      Red   x,y:        0.651, 0.332
      Green x,y:        0.317, 0.610
      Blue  x,y:        0.152, 0.069
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 30 ae a6 61 01 01 01 01   ........0..a....
      +0010   16 1b 01 03 a0 35 1e 78 2a c5 35 a6 55 51 9c 27   .....5.x*.5.UQ.'
      +0020   11 4e 54 af ef 00 d1 c0 81 80 81 8a 95 00 a9 c0   .NT.............
      +0030   a9 cf b3 00 71 4f 02 3a 80 18 71 38 2d 40 58 2c   ....qO.:..q8-@X,
      +0040   45 00 0f 28 21 00 00 1e 00 00 00 ff 00 56 54 35   E..(!........VT5
      +0050   31 38 31 36 32 0a 20 20 20 20 00 00 00 fd 00 32   18162.    .....2
      +0060   4b 0f 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   K.S...      ....
      +0070   00 54 32 34 69 2d 31 30 0a 20 20 20 20 20 01 d1   .T24i-10.     ..
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA, DDCRC_READ_ALL_ZERO(9)])

Display 1
   I2C bus:  /dev/i2c-10
      DRM connector:                         card1-DP-2
      /sys/class/drm/card1-DP-2/dpms         On
      /sys/class/drm/card1-DP-2/enabled      enabled
      /sys/class/drm/card1-DP-2/status       connected
      Driver:                                amdgpu
      I2C address 0x50 (EDID) responsive:    true 
      I2C address 0x37 (DDC)  responsive:    true 
[... 21 lines cut ...]
      Green x,y:        0.262, 0.662
      Blue  x,y:        0.149, 0.058
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 4c 2d 1d 0f 54 53 57 43   ........L-..TSWC
      +0010   32 30 01 04 a5 50 21 78 3a 02 75 ae 50 43 a9 26   20...P!x:.u.PC.&
      +0020   0e 50 54 bf ef 80 71 4f 81 00 81 c0 81 80 95 00   .PT...qO........
      +0030   a9 c0 b3 00 01 01 e7 7c 70 a0 d0 a0 29 50 30 20   .......|p...)P0 
      +0040   3a 00 1d 4d 31 00 00 1a 00 00 00 fd 00 32 64 1e   :..M1........2d.
      +0050   98 37 00 0a 20 20 20 20 20 20 00 00 00 fc 00 43   .7..      .....C
      +0060   33 34 4a 37 39 78 0a 20 20 20 20 20 00 00 00 ff   34J79x.     ....
      +0070   00 48 34 5a 4e 35 30 31 38 36 31 0a 20 20 01 4a   .H4ZN501861.  .J
   VCP version:         2.1
DDC: Unexpected VCP opcode 0xdf, should be 0xc8, response data bytes: 0200df00 ffff0201
DDC: Unexpected VCP opcode 0xdf, should be 0xc8, response data bytes: 0200df00 ffff0201
   Controller mfg:      Novatek
DDC: Unexpected VCP opcode 0xc8, should be 0xc9, response data bytes: 0200c800 ffff0012
DDC: Unexpected VCP opcode 0xc8, should be 0xc9, response data bytes: 0200c800 ffff0012
   Firmware version:    1.32
   Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
   Current dynamic sleep adjustment multiplier:  0.00

```

ojaskavathe, 2024-12-03:
I'm getting basically the same output on my BenQ GW2480 on running `ddcutil detect --verbose --ddcdata`:

```
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
Invalid display
   I2C bus:  /dev/i2c-7
      DRM connector:                         card2-HDMI-A-1
      /sys/class/drm/card2-HDMI-A-1/dpms     On
      /sys/class/drm/card2-HDMI-A-1/enabled  enabled
      /sys/class/drm/card2-HDMI-A-1/status   connected
      Driver:                                amdgpu
      I2C address 0x50 (EDID) responsive:    true
      I2C address 0x37 (DDC)  responsive:    true
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-7/name        AMDGPU DM i2c hw bus 1
      PCI device path:                       /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.0/i2c-7
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQ GW2480
      Product code:         30951  (0x78e7)
      Serial number:        22M0022501Q
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2021,  Week: 5
      EDID version:         1.3
      Extra descriptor:
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.652, 0.336
      Green x,y:        0.321, 0.610
      Blue  x,y:        0.153, 0.061
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 09 d1 e7 78 45 54 00 00   ...........xET..
      +0010   05 1f 01 03 80 35 1e 78 2e 05 65 a7 56 52 9c 27   .....5.x..e.VR.'
      +0020   0f 50 54 a5 6b 80 d1 c0 b3 00 a9 c0 81 80 81 00   .PT.k...........
      +0030   81 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 0f 28 21 00 00 1e 00 00 00 ff 00 32 32 4d   E..(!........22M
      +0050   30 30 32 32 35 30 31 51 0a 20 00 00 00 fd 00 32   0022501Q. .....2
      +0060   4c 1e 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   L.S...      ....
      +0070   00 42 65 6e 51 20 47 57 32 34 38 30 0a 20 01 d2   .BenQ GW2480. ..
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(10)])
```