rockowitz/ddcutil #468: Display flickers in brightness when adjusting brightness

adlr, 2024-11-10:
Fedora 41 on Lenovo X1 Carbon Gen 9 connected to LG UltraFine 27" 5K display. Extension is functional, but when adjusting brightness more than one "step," I'm seeing the display get brighter/dimmer/brighter/dimmer as it approaches the final correct brightness. This makes it challenging to adjust as it interferes with the human feedback loop.

PR incoming to address this.

adlr, 2024-11-10:
Apologies, opened this on the wrong project. Oops!

closed 2024-11-10 by adlr

rockowitz, 2024-11-12:
I'm relieved to know that the issue is not with ddcutil.  However I'm curious to know whether ddcui exhibits the same problematic behavior

adlr, 2024-11-12:
I managed to get ddcui installed, but I'm not sure how to use it to adjust
brightness. Any quick pointers? I'll be happy to do a quick test. Thanks!

rockowitz, 2024-11-14:
When ddcui starts, you will be in the Monitor Summary view.  In the Display drop-down, pick a display that's not listed as Invalid in the summary view (typically a laptop that does not support DDC/CI.
Select the Features view.  Feature x10 is brightness.  You can adjust the brightness using the slider or enter the spin box.

adlr, 2024-11-14:
Hm, I'm getting a crash when I try to go to the Features view:

```
ddcui: api_metadata.c:245: ddca_get_feature_list_by_dref: Assertion `vcp_version_is_valid(vspec, false)' failed.
zsh: IOT instruction (core dumped)  ddcui
```

I'm attaching the output from `ddcutil interrogate >lg_ultrafine.log 2>&1`
[lg_ultrafine.log](https://github.com/user-attachments/files/17742733/lg_ultrafine.log)

rockowitz, 2024-11-18:
You've hit an edge case I hadn't considered.  If you look at the output of **ddcutil detect**, you'll see " VCP version:         Detection failed".   And if you look at the "Scanning all VCP feature codes for display" section towards the end of the interrogate output, you'll see the line:
~~~
VCP code 0xdf (VCP Version                   ): Unsupported feature code
~~~ 

The **ddcutil** command itself does something reasonable in this situation, but the metadata portion of the API does not.  It needs to more gracefully handle monitors for which VCP version detection fails.

Thank you very much of this report.  Another example of bug around for years that hadn't been hit, or at least reported.