rockowitz/ddcutil #470: DDCA_EVENT_DPMS_AWAKE sent shortly after DDCA_EVENT_DPMS_ASLEEP

wojnilowicz, 2024-11-12:
My external screen goes to sleep correctly, but soon, after it displays "stand by" message by itself, it wakes up. It happens in KDE Plasma 6.

My display is a single Lenovo P24h-2L connected through HDMI to my laptop. I have ddcutil 2.1.2 and using nouveau on a NV50 era hardware. I can provide more details, if requested.

I described the issue in details in a [KDE bug #495746](https://bugs.kde.org/show_bug.cgi?id=495746). I'm reporting it also here because I see that DDCA_EVENT_DPMS events come from ddcutil. I'm not sure, if it's a bug in ddcutil or KDE, so feel free to close immediately.

rockowitz, 2024-11-18:
If you can, please build from branch 2.1.5-dev and retry.   Having **ddcutil** report  sleep/wake events as they happen proved problematic and has been removed from the API.  So the KDE should not emit a message that causes the monitor to wake.

wojnilowicz, 2024-11-19:
Indeed, using 2.1.5-dev at cae9f52c06769fe0636cb784021fddaa39121992 fixes my issue. Thanks.

Any particular commit that fixed it?

I think this can be closed anyway.

wojnilowicz, 2024-11-20:
I just discovered that I cannot have both:
1. display dimming 
2. display sleep
enabled at the same time, because the same issue returns even with 2.1.5-dev. 

Here is what happens:
1. no my activity for 30 s
2. my display dims to 80%
3. no my activity for another 30 s
4. my display goes to sleep
5. not more than five seconds passes
6. my display wakes up dimmed to 80%

Could it be also attributed to ddcutil?
In any case, I'm attaching the output of ```journalctl -b --user-unit plasma-powerdevil```
[powerdevil.txt](https://github.com/user-attachments/files/17835596/powerdevil.txt)

wojnilowicz, 2024-11-23:
It works as expected on today's 2.1.5-dev. Thanks again.

Will you make --disable-watch-displays enabled by default?

rockowitz, 2024-11-23:
Option ***--disable-watch-displays*** will not be enabled by default.  It is intended as workaround for a bug whose cause is unclear, and to help diagnose the locus of the problem.  It completely overrides the ability of PowerDevil to request that libddcutil watch for and report display connection changes.