rockowitz/ddcutil #472: ddcutil pi 4, Display not found. Cant use any command. Works perfectly fine on the same monitor with pi 3 and ddcutil.

olivertd, 2024-12-04:
Ive tried to do as the documentation says:;
Raspberry Pi 4

As of firmware 5.10, the HDMI interface on the Raspberry Pi 4 supports I2C. The following steps will enable ddcutil.

Ensure you are at least on firmware 5.10, which is now mainstream. If necessary, execute:
$ sudo apt update && sudo apt upgrade
In /boot/config/txt, look for the [Pi4] section and replace dtoverlay=vc4-fkms-v3d with dtoverlay=vc4-kms-v3d
In /etc/modules, add i2c_dev
Reboot
If you have not yet installed ddcutil: $ sudo apt install ddcutil
The I2C buses for the two HDMI ports are normally /dev/i2c-11 and /dev/i2c-12. However, the bus names will be /dev/i2c-12 and /dev/i2c-13 if  dtparam=i2c_vc_on is also present in config.txt.
Thank you Robert Alexa for updating the status of ddcutil on the Pi 4 and providing the above instructions. For a long thread on the problem of I2C support on the Pi 4, see [Raspberry Pi issue #3152](https://github.com/raspberrypi/linux/issues/3152#)

Update 6/2023

Fresh install of Rasberry Pi OS (64 bit), build 2023-05-03 a spin of Debian 11 (bullseye) with kernel 6.1 on a Raspberry Pi Model B. Enable I2C in raspi-config, and reboot. apt install ddcutil, installs 0.9.9. From OBS, installs 1.4.1.

When i do ddcutil detect it says invalid display, i2c bus: /dev/12c-21 etc etc and ends with DDC Communication failed. Why is this? The same monitor worked to use ddc from the pi 3 and a mini pc running ubuntu

rockowitz, 2024-12-05:
Try building ddcutil from branch 2.2.0-dev.  Several changes related to the Raspberry Pi went in a week and a half ago and it's now working for me.

olivertd, 2024-12-05:
> Try building ddcutil from branch 2.2.0-dev. Several changes related to the Raspberry Pi went in a week and a half ago and it's now working for me on a Pi 4.

Great, Ill try it thanks. Didnt realease the apt version was 1.4.

olivertd, 2024-12-05:
> Try building ddcutil from branch 2.2.0-dev. Several changes related to the Raspberry Pi went in a week and a half ago and it's now working for me on a Pi 4.

Same issue now as before with the 1.4.1 version, really weird since i saw that ddcutil worked for a lot of people now and only had this issue on a very old raspberry pi os version.

`pi@raspberrypi:~ $ ddcutil --version
ddcutil 2.2.0-dev
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

pi@raspberrypi:~ $ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-21
   DRM connector:           card1-HDMI-A-2
   EDID synopsis:
      Mfg id:               PKB - UNK
      Model:                Maestro243DL
      Product code:         468  (0x01d4)
      Serial number:        D280C0064020
      Binary serial number: 50342974 (0x03002c3e)
      Manufacture year:     2010,  Week: 30
   DDC communication failed

pi@raspberrypi:~ $ ddcutil setvcp 10 10
Display not found

pi@raspberrypi:~ $ 
`

closed 2024-12-07 by olivertd