rockowitz/ddcutil #476: Benq EX3210U either non responsive or invalid feature flag in DDC reply packet.

evertvorster, 2024-12-10:
Hi there!

I've been seeing an issue with my external monitor. It's a fairly new and expensive Benq EX3210U monitor, alledgedly capable of 4K@144Hz in HDR. The best I have seen is 4K@120fps in HDR and SDR, but that might be a different discussion. 

This monitor is connected via a HDMI 2.1 cable to the HDMI port of my Asus ROG Strix Scar laptop. It is an Advanced Optimus MUX type laptop, and apparently the HDMI port is directly wired to the nVidia RTX 4090 that is the laptops dGPU. 

The operating system is Arch Linux, fully up to date. 
The issue that I am seeing I have reported to the KDE bug tracker. 
https://bugs.kde.org/show_bug.cgi?id=496631

The reason that I am here, is that there seems to be some inconsistency in the way that ddcutil is detecting this display. 

If the monitor is powered down, unplugged from power and the HDMI port, and then plugged in sudo ddcutil detect --verbose --ddc --sleep-multiplier 20 returns:
```
[evert@Evert ~]$ sudo ddcutil detect --verbose --ddc --sleep-multiplier 20
Invalid display
   I2C bus:  /dev/i2c-3
      DRM connector:                         card1-eDP-1
      /sys/class/drm/card1-eDP-1/dpms        On
      /sys/class/drm/card1-eDP-1/enabled     enabled
      /sys/class/drm/card1-eDP-1/status      connected
      /sys/class/drm/card1-eDP-1/connector_id93
      Driver:                                amdgpu
      EDID exists:                           true 
      I2C address 0x37 (DDC)  responsive:    false
      Is LVDS or EDP display:                false
      Is laptop display by EDID:             false
      Is laptop display:                     false
      /sys/bus/i2c/devices/i2c-3/name        AMDGPU DM i2c hw bus 0
      PCI device path:                       /sys/devices/pci0000:00/0000:00:08.1/0000:09:00.0/i2c-3
   EDID synopsis:
      Mfg id:               BOE - BOE
      Model:                NE173QHM-NZ2
      Product code:         2921  (0x0b69)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2022,  Week: 24
      EDID version:         1.4
      Extra descriptor:        BOE CQ
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.680, 0.320
      Green x,y:        0.265, 0.690
      Blue  x,y:        0.150, 0.060
      Extension blocks: 1
   EDID source: SYSFS
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 09 e5 69 0b 00 00 00 00   ..........i.....
      +0010   18 20 01 04 a5 26 15 78 03 0f 95 ae 52 43 b0 26   . ...&.x....RC.&
      +0020   0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 e2 67 00 b0 a0 a0 b4 50 30 20   .......g.....P0 
      +0040   36 00 7d d6 10 00 00 1a 00 00 00 fd 0c 3c f0 86   6.}..........<..
      +0050   86 6a 01 0a 20 20 20 20 20 20 00 00 00 fc 00 4e   .j..      .....N
      +0060   45 31 37 33 51 48 4d 2d 4e 5a 32 0a 00 00 00 fe   E173QHM-NZ2.....
      +0070   00 42 4f 45 20 43 51 0a 20 20 20 20 20 20 01 62   .BOE CQ.      .b
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)
   If the monitor's on screen display has a DDC/CI setting, check it is enabled.
   Monitor Model Id:  BOE-NE173QHM_NZ2-2921
   Feature definition file BOE-NE173QHM_NZ2-2921.mccs not found.

Display 1
   I2C bus:  /dev/i2c-17
      DRM connector:                         card0-HDMI-A-1
      /sys/class/drm/card0-HDMI-A-1/dpms     On
      /sys/class/drm/card0-HDMI-A-1/enabled  disabled
      /sys/class/drm/card0-HDMI-A-1/status   connected
[... 182 lines cut ...]
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 09 d1 a6 7f 01 01 01 01   ................
      +0010   15 20 01 03 80 47 28 78 2a 75 65 b0 4e 2e bc 26   . ...G(x*ue.N..&
      +0020   0e 50 54 a5 6b 80 d1 c0 81 c0 81 00 81 80 a9 c0   .PT.k...........
      +0030   b3 00 81 bc 61 7c 4d d0 00 a0 f0 70 3e 80 30 20   ....a|M....p>.0 
      +0040   35 00 c4 8f 21 00 00 1a 00 00 00 ff 00 45 54 57   5...!........ETW
      +0050   35 4e 30 35 30 32 36 53 4c 30 00 00 00 fd 00 30   5N05026SL0.....0
      +0060   78 1e ff 77 00 0a 20 20 20 20 20 20 00 00 00 fc   x..w..      ....
      +0070   00 42 65 6e 51 20 45 58 33 32 31 30 55 0a 01 11   .BenQ EX3210U...
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)
   If the monitor's on screen display has a DDC/CI setting, check it is enabled.
   Monitor Model Id:  BNQ-BenQ_EX3210U-32678
   Feature definition file BNQ-BenQ_EX3210U-32678.mccs not found.
```

This may or may not be related to my original issue of KDE not outputting to this monitor if it wakes up with a defined frame rate higher than 30fps, but it is something that is a little interesting to me. 

For completeness, I am also attaching the output of sudo ddcutil interrogate when the monitor is freshly woken up and actually responds to ddcutil. It seems to hang on my system, so it might be incomplete. 

[2024-12-10_DDCUtil_interrogate_Fresh_Powerup.txt](https://github.com/user-attachments/files/18075746/2024-12-10_DDCUtil_interrogate_Fresh_Powerup.txt)

Any help in figuring out what is going on and which direction to look would be awesome. I have been in contact with BenQ to see if I can get a firmware update for this monitor, but so far we have been unable to update this monitor.

evertvorster, 2024-12-12:
Thanks, I have been looking at this for a while. 
I included all the information for completeness, but I do not expect you to troubleshoot my KDE woes. There is plenty wrong there, and I plan to tackle that once I don't see any errors coming from ddcutil and this monitor. 

I have, indeed got POWERDEVIL_NO_DDCUTIL=1 in /etc/environment. Of course, I don't really like that, because I would like ddcutil to properly detect and manage this display. Which is the reason we are poking at it now, I guess. 

>In the detect output for the second case where the computer is restarted with the monitor connected, the telling line in the output is "I2C address 0x37 (DDC) responsive: false". In that situation, issue command **i2cdetect -y 17", which shows which slave addresses are active on the I2C bus. Slave address x50 (EDID) should be reported. If slave address x37 (DDC) is not, the problem is entirely in the province of the driver and monitor, ddcutil has no involvement.

So, I have started up the computer with the monitor powered up, and indeed it is showing the laptop's panel on /dev/i2c-3 and /dev/i2c-7
I do have a docking station, but it is not connected at this time. Whether connected to the docking station, or to the HDMI port of the laptop, I see similar issues. 

The Benq shows up on /dev/i2c-17, and claims to be unresponsive. 

```
[evert@Evert ~]$ i2cdetect -y 17
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- 3e -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- 54 -- -- -- 58 -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
I don't know what the above means, hopefully you can shed some light on that? 

Valgrind is completely not happy:
```
[evert@Evert ~]$ valgrind ddcutil interrogate
==4070== Memcheck, a memory error detector
==4070== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==4070== Using Valgrind-3.24.0 and LibVEX; rerun with -h for copyright info
==4070== Command: ddcutil interrogate
==4070== 
vex amd64->IR: unhandled instruction bytes: 0x62 0xF3 0xF5 0x28 0x38 0xF0 0x1 0xC4 0xE1 0xF9
vex amd64->IR:   REX=0 REX.W=0 REX.R=0 REX.X=0 REX.B=0
vex amd64->IR:   VEX=0 VEX.L=0 VEX.nVVVV=0x0 ESC=NONE
vex amd64->IR:   PFX.66=0 PFX.F2=0 PFX.F3=0
==4070== valgrind: Unrecognised instruction at address 0x194d71.
==4070==    at 0x194D71: parse_command (in /usr/bin/ddcutil)
==4070==    by 0x148254: main (in /usr/bin/ddcutil)
==4070== Your program just tried to execute an instruction that Valgrind
==4070== did not recognise.  There are two possible reasons for this.
==4070== 1. Your program has a bug and erroneously jumped to a non-code
==4070==    location.  If you are running Memcheck and you just saw a
==4070==    warning about a bad jump, it's probably your program's fault.
==4070== 2. The instruction is legitimate but Valgrind doesn't handle it,
==4070==    i.e. it's Valgrind's fault.  If you think this is the case or
==4070==    you are not sure, please let us know and we'll try to fix it.
==4070== Either way, Valgrind will now raise a SIGILL signal which will
==4070== probably kill your program.
==4070== 
==4070== Process terminating with default action of signal 4 (SIGILL): dumping core
==4070==  Illegal opcode at address 0x194D71
==4070==    at 0x194D71: parse_command (in /usr/bin/ddcutil)
==4070==    by 0x148254: main (in /usr/bin/ddcutil)
==4070== 
==4070== HEAP SUMMARY:
==4070==     in use at exit: 39,395 bytes in 411 blocks
==4070==   total heap usage: 455 allocs, 44 frees, 54,675 bytes allocated
==4070== 
==4070== LEAK SUMMARY:
==4070==    definitely lost: 0 bytes in 0 blocks
==4070==    indirectly lost: 0 bytes in 0 blocks
==4070==      possibly lost: 0 bytes in 0 blocks
==4070==    still reachable: 39,395 bytes in 411 blocks
==4070==         suppressed: 0 bytes in 0 blocks
==4070== Rerun with --leak-check=full to see details of leaked memory
==4070== 
==4070== For lists of detected and suppressed errors, rerun with: -s
==4070== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Illegal instruction (core dumped)
[evert@Evert ~]$ 
```

In Arch, I don't see a package with "asan" in the name. Could you please provide a link to the source of that package so I can hunt it down?

evertvorster, 2024-12-12:
I did manage to get a full interrogation. I guess I was just not patient enough last time. 

[2024-12-12_DDCUtil_interrogate_fresh_powerup_complete.txt](https://github.com/user-attachments/files/18105668/2024-12-12_DDCUtil_interrogate_fresh_powerup_complete.txt)

For completeness, I have also run a ddcutil interrogate after a warm boot:
[2024-12-12_DDCUtil_Warm_boot_interrogate.txt](https://github.com/user-attachments/files/18105697/2024-12-12_DDCUtil_Warm_boot_interrogate.txt)

So, after a warm boot with the monitor on, this monitor now claims to not support DDC. Also, the remote for it does not work. In the past I have noticed some very weird issues, where it would flicker heavily at high refresh rates. Even turning off and unplugging everything and starting up again this flickering continued until I started playing a bit of No Mans Sky, when it stopped. This leads me to believe that there is something not working properly and giving the monitor signals which completely confuses it. 

One other thing that I notice with this monitor is that it is very slow to display an image once it is plugged in and has a valid HDMI signal sent to it.

evertvorster, 2024-12-12:
I'm starting to suspect a driver issue. If there is some way to completely rule out ddcutil, then I can follow the rabbit hole a bit deeper and go pester the folks at nVidia.

rockowitz, 2024-12-12:
Re the docking station, I was just pointing to that as another situation in  which the monitor appears on multiple /dev/i2c devices.  I have a hunch that, and it's only that, having 2 gpus in the system uses some of the same logic as docking stations.

**i2cdetect** is s a basic system utility independent of **ddcutil**.  It comes from package i2c-tools.  Command **i2cdetect -y 17** scans /dev/i2c-17 to see which slave addresses are responsive.  In this case it's slave addresses x3A, x3E, x50, x54, and x58. Slave address x50 is used to read the EDID; since it's responsive we know that there's a monitor connected.  Slave address x37 is used for DDC/CI communication; since it's not responsive DDC/CI communication with the monitor is impossible.  There's nothing ddcutil can do about it.  It's a driver/monitor issue.

Re valgrind, I've never seen output like that. but since the problem was imply that **ddcutil interrogate** was not allowed to run to completion, there's no reason to explore that more.  Re asan, FWIW on the Fedora system where I'm current;y running the package name is "libasan".

evertvorster, 2024-12-12:
Hi there!

Thank you so much for coming back to me on this issue. 

In response to the DDC/CI support disappearing when the monitor is connected and powered upon boot, I have filed this on the nVidia forums. 
https://forums.developer.nvidia.com/t/external-monitor-inconsistent-ddc-ci-detection/316597

On Arch libasan is part of gcc-libs, and installed on any Arch system that has GCC installed. 

We can almost wrap this up, but one curiosity remains from my side. 

When you look at the ddcutil interrogate output that was obtained with the monitor was adressable, there are a lot of VCP codes that are listed as unsupported VCP codes. Does this mean that the code is not implemented on this monitor, or that it is implemented on this monitor and ddcutil just does now know what to do with it?

In fact, when you scan through 
[2024-12-12_DDCUtil_interrogate_fresh_powerup_complete.txt](https://github.com/user-attachments/files/18107386/2024-12-12_DDCUtil_interrogate_fresh_powerup_complete.txt)
 is there anything that stands out to you about this monitor? 
Worded another way, in your opinion, is this more likely a monitor bug, or a driver bug?

evertvorster, 2024-12-12:
Thanks again for the prompt and very detailed response. 

I'll see if I can pester the manufacturer of the monitor to give me some documentation on what those codes mean, and I'll post it here if you can find some use for it.