rockowitz/ddcutil #477: GIGABYTE-M27Q change inout mode

10Nico17, 2024-12-13:
Hello,

I am having an issue with my GIGABYTE-M27Q monitor. While I can adjust other settings like brightness without any problems, I am unable to change the input mode.

Here are some details and screenshots to illustrate the issue:

[image]

[image]

command to change:
[image]

Is there a way to figure out what's causing this issue or a solution to resolve it?

Thank you for your help!

rockowitz, 2024-12-13:
Does the monitor actually switch to HDMI-1?  If not, the monitor probably uses a non-standard value for HDMI-1. It's a matter of trial and error to discover what the correct value is.

If the monitor does use non-standard values for feature x60, you can put that information in a User Defined Features file.

10Nico17, 2024-12-16:
Thank you very much for your answer. Unfortunately the monitor does not switch when I execute the command.
If I switch the input mode via the buttons on the monitor (which I wanted to do via the command in the future) it works and I can see with the getvcp command that hex numbers are correct for x60 (so 0x11 is HDMI1). Unfortunately the monitor does not switch when I do this with the setvcp command.

rockowitz, 2024-12-16:
I'm afraid I have nothing more to suggest, other than poking around.  Some monitors are just ornery.

closed 2024-12-17 by 10Nico17

1000001101000, 2025-05-16:
@10Nico17 is that a Gigabyte monitor with the KVM switch on the back?

I can switch inputs on mine (GIGABYTE-M32U) via x60, the values are:
```
0x0F: DisplayPort-1
0x10: USB-C
0x11: HDMI-1
0x12: HDMI-2
```

That can cause some issues with the attached mouse and keyboard also switching. To trigger the "KVM Switch" that Gigabyte's software or the button on the back use you need to send some custom feature codes that I'm having trouble figuring out how to send with ddcutil. It basically takes the format of a two-byte feature code (0E 69) which you send the value "1".

I've been triggering it with some python code I wrote a while back, the raw data looks like this:

SEND:  ['0x51', '0x85', '0x3', **'0xe0', '0x69',** '0x0', '0x1', '0x31']

for comparison, here is  changing to HDMI-1 via vcp 60

SEND:  ['0x51', '0x84', '0x3', **'0x60',** '0x0', '0x11', '0xc9']

@rockowitz is there a way to send 2-byte feature codes like this via `ddcutil`? Although writing the python code was fun I'd much rather put together udf/mccs file to use/publish if that option exists.

Thank you for your attention!

rockowitz, 2025-05-16:
Command **setvcp** accepts 2 byte values.  For **getvcp x60** you may want to use option ***--verbose*** to see the high byte (SH).  User defined feature type XNC was created to address this situation.  See section [User Defined Features](https://www.ddcutil.com/release_notes_220/#user-defined-features) in the Release Notes for release 2.2.0.

1000001101000, 2025-05-16:
Looking at XNC and UDF files, it seems like I would need 0E69 to be a valid feature code (I don't think the spec works like that?) which would allow setvcp to insert the 0x1/write byte between that feature code and the value 0x1.

Alternatively, I could try to add the  0x1/write byte as part of the value but that would require >2-byte values which seem to be unsupported for XNC. 
```
Number must be in range 0..65535:  6881281
Invalid VCP value: 0x690001
```

Looking at the spec again (it's been a few years), this may need to be a Table value rather than C/NC. 
```
1.4.3 VCP Code Function
The ‘Function’ column in Table 8-2, Table 8-4, Table 8-6, Table 8-8, Table 8-9, Table 8-11, Table
8-13, Table 8-15, Table 8-17, and Table 8-18 refer to the permissible action(s) with each VCP code:
  C : Continuous
  NC : Non-continuos
  T : Table
```
Is there any option in ddcutil for specifying custom table functions? 

here is the mccs file I made as part of the attempt
```
MFG_ID       GBT
MODEL        Gigabyte M32U
PRODUCT_CODE 12804
MCCS_VERSION 2.2
FEATURE_CODE E0 Manuf Custom
    ATTRS XNC RW
```

rockowitz, 2025-05-17:
@1000001101000  I apologize for sending you down the XNC rabbit hole. I was thinking about 2 byte values.  Feature type XNC is for reporting a non-continuous feature value whose low byte (SL) is one of a list of values as is common, but whose high byte (SH) also has meaning.  It's irrelevant for what you're trying to do.  

There's no way to specify a two byte feature code in ddcutil.  It's simply outside the DDC/CI spec.  

If `SEND: ['0x51', '0x85', '0x3', '0xe0', '0x69', '0x0', '0x1', '0x31]` is indeed what works, my best interpretation is that feature code xE0 is sending a 3 byte value.  This is also is outside the spec - the Set VCP Feature Value packet can only send 2 bytes.  (See section 4.4 Set Vcp Feature of Display Data Channel Command Interface Standard Version 1.1.)  

Nor can what you'd like to do be put into a Table Write packet.   In that case the op code (the 3rd byte of what is sent) would be xE7 for Table Write, not x03 for Set VCP Feature. 

I'm afraid you're stuck with your Python code.