rockowitz/ddcutil #479: Samsung LS27D70xE - Setting value failed for feature x10

st0rm-shad0w, 2024-12-15:
Hello, I can't get ddcutil to run to change the brightness on my new monitor that is connected to a desktop box using an HDMI cable.

Error
```
Setting value failed for feature x10, rc=DDCRC_RETRIES(-3007): maximum retries exceeded
    Try errors: DDCRC_RETRIES
```
HW Details:
```
ddcutil detect
Display 1
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                LS27D70xE
      Product code:         30347  (0x768b)
      Serial number:        HNAX601157
      Binary serial number: 810899530 (0x3055584a)
      Manufacture year:     2024,  Week: 26
   VCP version:         2.0.
```

The result of `ddcutil interrogate` is attached.

[interrogate.txt](https://github.com/user-attachments/files/18141515/interrogate.txt)
[ddcdata.txt](https://github.com/user-attachments/files/18141512/ddcdata.txt)

rockowitz, 2024-12-15:
Please build from branch 2.2.0-dev, run the following, and submit the output as attachments

ddcutil setvcp 10 20 --trcfunc ddc_write_read_with_retry --trcfunc ddc_write_only_with_retry
ddcutil setvcp 10 20 --trcfunc ddc_write_read_with_retry --trcfunc ddc_write_only_with_retry --noverify
ddcutil getvcp 10 --trcfunc ddc_write_read_with_retry --trcfunc ddc_write_only_with_retry --noverify

st0rm-shad0w, 2024-12-15:
Built and installed 2.2.0-dev HEAD

```
stormshadow@cobracommand ~> ddcutil -V
ddcutil 
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Output for the requested commands.

[setvcp.txt](https://github.com/user-attachments/files/18142144/setvcp.txt)
[setvcp-noverify.txt](https://github.com/user-attachments/files/18142143/setvcp-noverify.txt)
[getvcp-noverify.txt](https://github.com/user-attachments/files/18142146/getvcp-noverify.txt)

rockowitz, 2024-12-16:
The sleep-multipliers being used are very low.  Try explicitly setting option ***--sleep-multiplier 2.0***.

rockowitz, 2024-12-16:
Also, is the returned getvcp value for feature x10 really correct?  The display would have to be dark or at least very dim.  What happens if you adjust brightness in the screen's OSD?  Is that even possible?

Feature xdc reports that the monitor display mode is x02 - Mixed.  It's possible that the mode disallows changing brightness.  Try setting the display into a User mode in the OSD.

st0rm-shad0w, 2024-12-16:
> Also, is the returned getvcp value for feature x10 really correct? The display would have to be dark or at least very dim. What happens if you adjust brightness in the screen's OSD? Is that even possible?

Well the value noted in getvcp matches screen's OSD. But it doesn't get dark/ very dim at lower values (0). On the higher value side it only gets so bright eyes hurt. 😅

> Feature xdc reports that the monitor display mode is x02 - Mixed. It's possible that the mode disallows changing brightness. Try setting the display into a User mode in the OSD.

This was the issue. I had adaptive brightness/ eye care technology on. After disabling it from the OSD, I'm able to change the brightness with ddcutil. Thanks!

Actually got the same hint here - https://wiki.archlinux.org/title/Backlight

> ddcutil will fail to set some VCP features if there is a feature enabled on the monitor which already automatically adjusts them (e.g. [Dynamic Contrast Ratio](https://en.wikipedia.org/wiki/Contrast_ratio#Dynamic_contrast_.28DC.29) or BenQ's Eye Care technology).

Maybe having it mentioned in the top-level README would be a good inclusion?

st0rm-shad0w, 2024-12-16:
> The sleep-multipliers being used are very low. Try explicitly setting option _**--sleep-multiplier 2.0**_.

What does this do?

st0rm-shad0w, 2024-12-16:
Somehow I wasn't able to spot it. But if it's there, it should be fine, I suppose.

Also, a few other things noted:

1. ddcutil 2.14 works too, just needs the brightness/ eye care technology to be disabled. However the brightness change transition seems to be smoother in 2.20.
2. Even after disabling brightness/ eye care technology, the display mode is x02 - Mixed. Wonder how to change display into User mode, there are no explicit options in the OSD.

st0rm-shad0w, 2024-12-16:
Also, thanks again for all the help. Please feel free to close the issue.

rockowitz, 2024-12-16:
Try setting feature x14, color preset.  According to the **CAPABILITIES** command, which reads and interprets the monitor's  capabilities string, it implements values x05, x08, x0b, and X0c, which are 6500K, 9300K, User 1, and User 2 respectively.  The interaction between feature x14 (color preset) and feature xdc (display mode) is not defined in the Monitor Control Command Set spec - so it's a matter of how the monitor maker chose to implement the interaction.  

And note that while the monitor responds to a request to read feature xdc, it is not reported in the capabilities string.  Capabilities strings are regarded as hints, not taken as gospel.   To see all the information that **ddcutil** has for a feature per the MCCS spec, use **ddcutil capabilities <your feature code here> --verbose**.

st0rm-shad0w, 2024-12-17:
I spoke too soon! After a shutdown and power-up, the issue reoccurred, and nothing seemed to fix it despite trying several solutions.

However, toggling the "Brightness/Eye Care Technology" setting from ON to OFF somehow restored functionality to ddcutil, as if it re-authenticated or reset something.

Additionally, I was unable to set up feature X14 both before and after this issue.