rockowitz/ddcutil #480: Contrast changes the display but it's not "real" for the OSD

Theluga, 2024-12-22:
Interrogate output is here: 
https://privatebin.arch-linux.cz/?bae0a4c27eaf96ec#F6gXWJUXE1T5Vwv8EUnF7NMSmXnoDq5Y9WvgQKcJZLBD

Hello! I have a Monitor from AOC model 22B1HM5-22B2HM.

I can edit the contrast by:

`ddcutil --sn=WXDO4XA004311 setvcp 0x12 value`

And it will work, but on the monitor OSD will display the previous value and the `getvcp` too will display the previous value and the command from set will also output:

 `Verification failed for feature 12` 

But the screen changes. I can see the effect of the changing of contrast. I don't know if it's a bug of the monitor or of the ddcutil.

--- 

And a little off-topic but somewhat similar. These codes will work but it will not change the value on the OSD (the monitor can only change all values together but the command can change individually, and when all are of the same value, the monitor does not update its OSD. I believe this is a monitor bug).

```
VCP code 0x70 (Video black level: Blue 
VCP code 0x6c (Video black level: Red
VCP code 0x6e (Video black level: Green
```

Why those codes are not shown in the capability command? 

Thank you for your great work and this great software!

rockowitz, 2025-01-02:
What you're seeing is consistent.  Verification performs a getvcp operation.  So it reads the same value as reported in the OSD.  

Your comment re feature codes x70, x6c, and x6e are consistent with this.  Again, the values are successfully set using DDC, but the new state is not being recorded, so the OSD is unchanged. 

A further comment regarding setting the RGB black levels, bear in mind that for some monitors, setting certain feature values (e.g. brightness) my be disabled depending on display mode (e.g. Cinema).

Theluga, 2025-01-02:
Can I say then that the monitor is bugged when changing the contrast?

 I believe this monitor has a program to control it from windows. But I don't use windows to know if contrast will be buggier there too.

rockowitz, 2025-01-15:
Yes.

closed 2025-01-30 by Theluga