rockowitz/ddcutil #481: Monitor not getting detected on suspend but is on resume

tarsobcaldas, 2024-12-27:
I'm trying to use ddcutil on a suspend script for systemd, as I did in other computers before, but the monitor is not getting detected when the computer is suspending, although it is when it resumes.

This is the script I'm running:

``` bash
#!/bin/sh
if [ "${1}" == "pre" ]; then
  ddcutil detect
  ddcutil environment
  ddcutil interrogate --verbose
  ddcutil setvcp D6 05  --bus 11 --disable-capabilities-cache --disable-dynamic-sleep
elif [ "${1}" == "post" ]; then
  ddcutil setvcp D6 01 --bus 11 --disable-capabilities-cache --disable-dynamic-sleep 
fi
```

And this is the output of `journalctl -b -u systemd-suspend.service`

```
Dec 27 17:08:46 sleipnir systemd-sleep[46699]: User sessions remain unfrozen on explicit request ($SYSTEMD_SLEEP_FREEZE_USER_SESSIONS=0).
Dec 27 17:08:46 sleipnir systemd-sleep[46699]: This is not recommended, and might result in unexpected behavior, particularly
Dec 27 17:08:46 sleipnir systemd-sleep[46699]: in suspend-then-hibernate operations or setups with encrypted home directories.
Dec 27 17:08:50 sleipnir systemd-sleep[46705]: Unable to determine dynamic sleep cache file name
Dec 27 17:08:50 sleipnir ddcutil[46705]: Unable to determine dynamic sleep cache file name
Dec 27 17:08:50 sleipnir systemd-sleep[46705]: Unable to determine dynamic sleep stats file name
Dec 27 17:08:50 sleipnir systemd-sleep[46705]: No displays found.
Dec 27 17:08:50 sleipnir systemd-sleep[46705]: Run "ddcutil environment" to check for system configuration problems.
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Unable to determine dynamic sleep stats file name
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: The following tests probe the runtime environment using multiple overlapping methods.
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: *** Basic System Information ***
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: ddcutil version: 2.1.4
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: /proc/version:
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Linux version 6.12.6-arch1-1 (linux@archlinux) (gcc (GCC) 14.2.1 20240910, GNU ld (GNU Binutils) 2.43.0) #1 SMP PREEMPT_DYNAMIC Thu, 19 Dec 2024 21:29:01 +0000
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Architecture:     x86_64
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Distributor id:   Arch
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Release:          rolling
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Found a known architecture
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: /proc/cmdline:
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    initrd=\initramfs-linux.img root=UUID="7549799f-7fdf-4c4f-a8fa-9d51900a643f" rw quiet loglevel=0 vt.globa_sursor_default=0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: *** Primary Check 1: Identify video card and driver ***
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Obtaining card and driver information from /sys...
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Primary video controller at PCI address 0000:0c:00.0 (boot_vga flag is not set)
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Device class:        x030000    VGA compatible controller
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Vendor:              x1002      Advanced Micro Devices, Inc. [AMD/ATI]
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Device:              x13c0      Granite Ridge [Radeon Graphics]
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Subvendor/Subdevice: 1043/8877  ASUSTeK Computer Inc.
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Driver name:         amdgpu
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Driver version:      Unable to determine
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-1      name: AMDGPU DM i2c hw bus 0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-2      name: AMDGPU DM i2c hw bus 1
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Primary video controller at PCI address 0000:01:00.0 (boot_vga flag is set)
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Device class:        x030000    VGA compatible controller
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Vendor:              x10de      NVIDIA Corporation
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Device:              x1f08      TU106 [GeForce RTX 2060 Rev. A]
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Subvendor/Subdevice: 10de/12fd  NVIDIA Corporation
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Driver name:         nvidia
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    Driver version:      565.77
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-10     name: NVIDIA i2c adapter 4 at 1:00.0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-8      name: NVIDIA i2c adapter 1 at 1:00.0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-11     name: NVIDIA i2c adapter 5 at 1:00.0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-9      name: NVIDIA i2c adapter 3 at 1:00.0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:    I2C device:          i2c-12     name: NVIDIA i2c adapter 6 at 1:00.0
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: *** Primary Check 2: Check that /dev/i2c-* exist and writable ***
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Current user: root (0)
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Checking /dev/i2c-* devices...
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Devices /dev/i2c-* must exist and the logged on user must have read/write permission for those devices (or at least those devices associated
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: with monitors).
Dec 27 17:08:54 sleipnir systemd-sleep[46739]: Typically, this access is enabled by:
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:   - setting the group for /dev/i2c-* to i2c
Dec 27 17:08:54 sleipnir systemd-sleep[46739]:   - setting group RW permissions for /dev/i2c-*
[... 67 lines cut ...]
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    loadable_i2c_dev_exists:       true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    i2c_dev_loaded_or_builtin:     true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    group_i2c_checked:             true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    group_i2c_exists:              true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    dev_i2c_common_group_name:     i2c
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    all_dev_i2c_has_group_i2c:     true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    any_dev_i2c_has_group_i2c:     true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    all_dev_i2c_is_group_rw:       true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    any_dev_i2c_is_group_rw:       true
Dec 27 17:10:25 sleipnir ddcutil[46739]: Unable to determine dynamic sleep cache file name
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    cur_uname:                     root
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    cur_uid:                       0
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    cur_user_in_group_i2c:         true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    cur_user_any_devi2c_rw:        true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    cur_user_all_devi2c_rw:        true
Dec 27 17:10:25 sleipnir systemd-sleep[46739]: Configuration suggestions:
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    Current user has RW access to all /dev/i2c-N devices.
Dec 27 17:10:25 sleipnir systemd-sleep[46739]:    Skipping further group and permission checks.
Dec 27 17:10:25 sleipnir ddcutil[46880]: dh=Display_Handle[i2c-11: fd=3], Keeping adjusted sleep multiplier 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
Dec 27 17:10:26 sleipnir systemd[1]: systemd-suspend.service: Deactivated successfully.
Dec 27 17:10:26 sleipnir systemd[1]: Finished System Suspend.
Dec 27 17:10:26 sleipnir systemd[1]: systemd-suspend.service: Consumed 1min 25.437s CPU time, 33.3M memory peak.
```

The command works fine executing it normally both as my user and as root.

rockowitz, 2025-01-02:
Per the Monitor Control Command Set (MCCS) spec, value x05 is a write-only value documented as "Write only value to turn off display".  Whether this actually completely turns off the display or puts it into standby mode is, I imagine, display specific.  Consider using `setvcp d5 2` to put the display into standby mode instead (assuming that 2 is a valid value for your monitor).

tarsobcaldas, 2025-01-03:
It does turn off the display, as I can do it from my user, and I also have the same configuration that works on my laptop, it's just not working when putting on suspend. Changing the command wouldn't work, because it's not recognizing the display when it's on the suspend process, although it recognizes when it's resuming.