rockowitz/ddcutil #482: RPI: Cannot control anymore HDMI display after action on feature D6 (power mode) - Invalid display - DDC communication failed

SimoneMSR, 2024-12-27:
I was correctly using my HDMI screen with ddcutil on a Raspberry Pi. Then I wanted to turn off the display. I run
```
ddcutil capabilities
[...]
Feature: D6 (Power mode)
  Values:
     01: DPM: On,  DPMS: Off
     04: DPM: Off, DPMS: Off
[...]
```
Then I run:

`ddcutil setvcp D6 04`

I see the display switch off. But then I run again:
```
$: ddcutil capabilities
Display not found
```
And I see this

```
$: ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-21
   DRM connector:           card1-HDMI-A-2
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                S24C31x
      Product code:         29457  (0x7311)
      Serial number:        
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2023,  Week: 48
   DDC communication failed
```

ddcutil environment gives this:
```

The following tests probe the runtime environment using multiple overlapping methods.
*** Basic System Information ***

ddcutil version: 1.4.1

/proc/version:
   Linux version 6.6.63-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1821 SMP PREEMPT Mon Nov 25 13:50:21 GMT 2024

Architecture:     aarch64
Distributor id:   Debian
Release:          12
Found a known architecture

/proc/cmdline:
   coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:EE:19:0F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=9803b9fb-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=IT

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Driver name:    vc4_v3d

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Current user: user (1000)

Checking /dev/i2c-* devices...

Devices /dev/i2c-* must exist and the logged on user must have read/write permission for those devices (or at least those devices associated 
with monitors).

Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c

Alternatively, this can be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89,  1 27 dic 21.21 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 20 27 dic 21.21 /dev/i2c-20
   crw-rw---- 1 root i2c 89, 21 27 dic 21.21 /dev/i2c-21

Current user (user) has RW access to all /dev/i2c-* devices.

*** Primary Check 3: Check that module i2c_dev is loaded ***

Checking for driver i2c_dev...
   sysfs reports module i2c_dev is loaded.
   Module i2c_dev is NOT built into the kernel
   Loadable i2c-dev module found
   Module i2c_dev is loaded

*** Primary Check 4: Driver specific checks ***

Performing driver specific checks...
No driver specific checks apply.

*** Additional probes ***

Examining /sys/bus/i2c/devices...
   /sys/bus/i2c/devices/i2c-1/name:   bcm2835 (i2c@7e804000)
   /sys/bus/i2c/devices/i2c-20/name:  fef04500.i2c
   /sys/bus/i2c/devices/i2c-21/name:  fef09500.i2c

Env_Accumulator:
   architecture:                  aarch64
   distributor_id                 Debian
   Drivers detected:              vc4_v3d
   /dev/i2c device numbers:       1 20 21
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   1 20 21
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     user
   cur_uid:                       1000
   cur_user_in_group_i2c:         true
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
```

Please. How can I control again my display with ddcutil?

SimoneMSR, 2024-12-30:
I solved by running the command:
Turn off
```
ddcutil setvcp D6 02 --noverify
```
Turn on
```
ddcutil setvcp D6 01
```
Initially, the `ddcutil setvcp D6 02` without the --noverify option, caused a brief turn off of the display, which suddenly re-turned on, each time.
I guess the `--noverify` option is the solution. Maybe the `setvcp` command without the --novery option asks for a confirmation to the device, so forcing it to wake up.

closed 2024-12-30 by SimoneMSR

rockowitz, 2025-01-02:
Re your initial post, after running `ddcutil setvcp d6 4`, the display is turned off.  However, depending on the display, the EDID may or may not be readable when the display is turned off.  In your case, it is readable, so `ddcutil detect` finds the display, but marks it invalid instead of assigning it a display number since DDC communication fails.  If no display number is specified on` ddcutil capabilities,` it defaults to display 1.  But in the current case there is no display 1, so it reports "Display not found". 

Re your solution, you've found if you put the display into standby (`ddcutil setvcp d6 2`) as opposed to turning it off (`ddcutil setvcp d6 4`), it remains responsive to ddcutil commands.  Again, whether a display responds to DDC commands when it is in standby mode depends on the display implementation; in your case luckily it does.

Finally, unless ***--noverify*** is specified, command **setvcp** performs a **getvcp** operation after setting the feature value to check that the value has been set.   That DDC reports success for the **setvcp** operation only means that the display has successfully received the request packet.  Given that this implicit **getvcp** operation wakes up the display, I suspect that any **getvcp** operation subsequent to the **setvcp** would wake it up.