rockowitz/ddcutil #483: Running detect causes my monitor's controls to lock up.

twoda5h, 2025-01-10:
```
$ ddcutil detect

Invalid display
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQ GW2490
      Product code:         30967  (0x78f7)
      Serial number:        [Redacted lol]
      Binary serial number: 1 (0x00000001)
      Manufacture year:     2024,  Week: 11
   DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_NULL_RESPONSE(10)])
```
after this runs, my monitor's power button, and all other buttons, cease to reply to presses. completely locked up until disconnected from power.

fun fact: my monitor does in fact have DDC...

rockowitz, 2025-01-10:
That the monitor locks up is disconcerting; it's not a behaviour I've seen.  Try using options ***--bus 6 --skip-ddc-checks*** to reduced the monitor interactions on the **detect** command. 

The DDC Null Message is a valid monitor reply request.  It indicates that monitor is unable to form a proper reply, though some monitors abuse the spec and use a DDC Null Message to indicate that the requested feature does not exist. 

If you're lucky, giving the monitor more time to  formulate a reply may help;  Use option ***--sleep-multiplier 2.0*** to cause ddcutil to double the time it waits between sending a request packet and attempting to read a response. 

Without **interrogate** output there's not much more I can say.

twoda5h, 2025-01-10:
> That the monitor locks up is disconcerting; it's not a behaviour I've seen. Try using options _**--bus 6 --skip-ddc-checks**_ to reduced the monitor interactions on the **detect** command.

```
$ ddcutil detect --bus 6 --skip-ddc-checks --sleep-multiplier 2.0

detect does not support explicit display option
```

> Without interrogate output there's not much more I can say.

the `interrogate` output is really long, how should i run it/send it?

rockowitz, 2025-01-10:
My oops.   ***--skip-ddc-checks*** is applicable to all commands, but ***--bus*** only applies to single display commands like **getvcp**.   Each reduce the overhead operations at the start of commands. 

To submit the **interrogate** output, probably simplest to click "Paste, drop, or click to add files" below the text entry area, or click on the paper clip above it.

twoda5h, 2025-01-11:
[here is the interrogate output](https://github.com/user-attachments/files/18383671/interro.txt)

twoda5h, 2025-01-11:
```
$ ddcutil detect --skip-ddc-checks

Display 1
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQ GW2490
      Product code:         30967  (0x78f7)
      Serial number:        [Redacted]
      Binary serial number: 1 (0x00000001)
      Manufacture year:     2024,  Week: 11
   VCP version:         Detection failed

```

rockowitz, 2025-01-19:
A final thought is that the monitor is in Eye-Care or some other mode such as Cinema that disables external control using DDC. (I assume you've actually verified that DDC is enabled in the OSD.) 

That  said, if ddcutil causes your monitor controls to lock up I'm afraid the only solution is to not use ddcutil.