rockowitz/ddcutil #487: Xiaomi Mi Monitor Curved 34" DDC communication failed when output is DP2

angelmixu, 2025-01-20:
Hi!

I was building a script to check for my USB KVM Switch and enabling one or the other input in my monitor.
I'm using a linux PC (connected to DP1) and a Windows 11 laptop (DP2) connected to this monitor.

In my linux PC, when using ddcutil, if the output is set to DP1, it shows every capability and can query for parameters properly.
When the monitor is set to DP2, showing my laptop content, it shows the following message:
`DDC communication failed. (getvcp of feature x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_NULL_RESPONSE(10)])
`

It's weird because when I'm in DP1 and send the message to change to DP2 the monitor goes blank for a moment, and goes back to DP1.

I attached both outputs from `sudo ddcutil interrogate --verbose`

[ddcutil-dp1-pc.txt](https://github.com/user-attachments/files/18482936/ddcutil-dp1-pc.txt)
[ddcutil-dp2-laptop.txt](https://github.com/user-attachments/files/18482937/ddcutil-dp2-laptop.txt)

Would you know what would happen?
Let me know if I can help in any way!

Thanks :)

rockowitz, 2025-01-24:
As you've probably noticed, the Xiaomi Mi is, to say the least, a very problematic monitor from a DDC perspective.   See issue #153.

The failure of DDC communication when sending requests on DP1 when the monitor input is set to DP2 is not surprising.  Whether a monitor responds to DDC requests from an input when that is not the current input is monitor dependent.

The capabilities output reports that the valid values for feature x60 Input source are x11, x12, x0f and x10.  The Monitor  Control Command Set (MCCS) spec defines these values as as HDMI-1, HDMI-2, DisplayPort-1 and DisplayPort-2 respectively.  However, the value reported for feature x60 is x07, which the spec defines as S-Video-1.  So the actual values used by the monitor are not those of the spec.   It's thus not surprising that if you've used command `ddcutil setvcp -x60 x10 --bus 5` trying to switch to DP-2 that the monitor does not switch input.  I'd suggest you try to determine what the actual feature values are by issuing commands `ddcutil setvcp x60 x01 --bus 5`, `ddcutil setvcp x60 x02 --bus 5` etc.

angelmixu, 2025-01-27:
Thanks @rockowitz !!!

I've tested from x00 to x34 values and none of them work :/
as in `ddcutil setvcp x60 x34 --bus 5`

Also, when using DP-2, and sending any command, it says:
`$ ddcutil setvcp x60 x24 --bus 5
DDC communication failed for monitor on bus /dev/i2c-5
`

And if I getvcp it shows this:
`$ ddcutil getvcp x60 --bus 5
VCP code 0x60 (Input Source                  ): S-Video-1 (sl=0x07)
`

Any other suggestion?

rockowitz, 2025-02-08:
Nothing more to suggest, I'm afraid.