rockowitz/ddcutil #490: Samsung LS32BG852NNXGO error DDCRC_RETRIES in ddc_write_read_with_retry, causes: EREMOTEIO(10)

ThracianKnight1907, 2025-02-01:
Hi, 

I'm not sure if this monitor supports DDC over Displayport. I checked in journal ctl erros relating to ddc and only found this. On Hdmi it works, but on displayport plasma seems to change brightness in software only (as in darkening the colors). The changes made are not transfered to the display. 

I have attached the output of `sudo ddcutil interrogate --verbose`
thank you

[ddcutil_neog8.txt](https://github.com/user-attachments/files/18628422/ddcutil_neog8.txt)

ThracianKnight1907, 2025-02-06:
could this be a fault of the cable? i'm still in the return window.

I tried asking Samsung customer support, but I don't think they know what DDC is

rockowitz, 2025-02-08:
There have been reports of DDC/CI being implemented on Samsung monitors only on the HDMI port but not the DP port.
See [this Reddit post](https://www.reddit.com/r/ultrawidemasterrace/comments/za2e2e/owners_of_a_samsung_odyssey_ultrawides_unite_we/)