rockowitz/ddcutil #491: Second Monitor Not Detected When Using Dock

Ragnoroct, 2025-02-05:
I installed `ddcutil` and I was successfully able to detect my HDMI monitor and change its brightness. It also detects my laptops builtin display but does not detect my secondary dell monitor.

**setup info**
```
os: Ubuntu 24.04.1 LTS (Noble Numbat)
kernel: 6.8.0-52-generic

dock: Dell Thunderbolt Dock - WD22TB4 (connected to thunderbolt port on laptop)
laptop: Dell XPS 13 9310

monitor primary:   cord_type=HDMI con=DP-7 model=DELL S2725DS
monitor secondary: cord_type=DP   con=DP-5 model=DELL S2725DS (not detected by ddcutil)

ddcutil version: 2.1.4 (installed using apt)
```

**ddcutil interrogate output**
[ddcutil-interrogate.log](https://github.com/user-attachments/files/18677435/ddcutil-interrogate.log)

**ddcutil detect output (missing secondary monitor GHTKR44)**
```
$ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-14
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               SHP - Sharp Corporation
      Model:                
      Product code:         5370  (0x14fa)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2020,  Week: 21
   This is a laptop display.  Laptop displays do not support DDC/CI

Phantom display
   Associated non-phantom display: bus /dev/i2c-20
   I2C bus:  /dev/i2c-15
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL S2725DS
      Product code:         53630  (0xd17e)
      Serial number:        1NTKR44
      Binary serial number: 808605762 (0x30325842)
      Manufacture year:     2024,  Week: 43
   DDC communication failed
   Use non-phantom device bus /dev/i2c-20

Display 1
   I2C bus:  /dev/i2c-20
   DRM connector:           card1-DP-7
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL S2725DS
      Product code:         53630  (0xd17e)
      Serial number:        1NTKR44
      Binary serial number: 808605762 (0x30325842)
      Manufacture year:     2024,  Week: 43
   VCP version:         2.1
```

rockowitz, 2025-02-07:
First, please clarify the hardware situation.   Are both external monitors connected  to the docking station, or is one connected to the dock and one directly to the laptop?

Ragnoroct, 2025-02-07:
Both are connected to the docking station, one using an hdmi and the other using displayport. The docking station is connected to the laptop through the thunderbolt connection.

rockowitz, 2025-02-08:
There are two things going on here.

The first, and the important one for you, is that DDC communication is not working.  I can see that in multiple places in the interrogate output.   In particular, the output of command **i2cdetect -y 15** shows that although slave address x50 (EDID) is responsive on /dev/i2c-15, slave address x37 (DDC) is not.   i2cdetect is a system utility command that is not part of ddcutil; ddcutil just uses it for diagnostic purposes.  So the problem is either in the docking station hardware, or the i915 driver.  I suggest you submit a bug report to Ubuntu, which will ultimately get to the i915 driver developers.

The second has to do with how ddcutil interprets the failure situation.  It is possible with docking stations (i.e. Multi-Stream Transport) that the same display appears on two different I2C buses. (For a long discussion, see i915 issue [Same MST display appears as 2 /dev/i2c devices](https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/3605#note_1317298)  In that case, ddcutil looks at the EDIDs read on each bus and, if they match, reports the one on which communication fails as a phantom display rather that invalid.  In this context that is purely informational.  You appear to have two monitors with identical identifying information including the serial number.  Is that really the case?

Ragnoroct, 2025-02-11:
>  You appear to have two monitors with identical identifying information including the serial number. Is that really the case?

No, the monitor with serial number `1NTKR44` is only one physical monitor. It is my main display and is actually the one ddcutil can change the brightness of.

My second physical monitor of the two has serial number `GHTKR44` which we don't see it in the describe output.

I opened a ticket for the package i2c-tools. Hopefully that was the right place.
https://bugs.launchpad.net/ubuntu/+source/i2c-tools/+bug/2098032

rockowitz, 2025-02-13:
I've taken another look at your problem, and it's stranger than I thought.   The problem on DP-5 is that the EDID, when read over DP, appears to have an unexpected length of 384 bytes, i.e. 3 128 byte blocks.  2 128 byte blocks is normal, the base block plus one extension block, 3 is unexpected.  Extension blocks are 128 bytes in length, and begin with x02.  The block beginning at offset 128 is normal, the block beginning at offset 256 is not.  Interestingly, though the monitors are the same model, the EDID presented from the HDMI connection is 256 bytes in length, which ddcutil knows how to handle.  So the i915 driver may have a hand in this situation, or it may be that ddcutil has to handle this case. My understanding is that only the base block and 1 extension block can be read at one time, with the extension block chosen by writing to I2C slave address x30.

Unfortunately, my focus right now is on getting release 2.2.0 out the door, and only show stopper bugs are being addressed.  However, there are several different paths through the code that can read the EDID in different ways.  If you're able to build and install from branch 2.2.0-dev, I have several suggestions with of options to try.  Perhaps one of them will finesse your situation.

***--edid-read-size 128*** only attempt to read the EDID as a 128 byte block
***--edid-read-size 256*** only attempt to read the EDID as a 256 byte block
***--f5*** uses special purpose code to read the EDID instead of reading the /dev/i2c device in the normal way
***--use-file-io*** calls into the i2c-dev driver in a different way; the default is --use-ioctl-io
***--enable-try-get-edid-from-sysfs*** tries to get the EDID from /sys before reading it directly; this may or may not work.  Depending on how the video driver has populated /sys, it may be that the only way to associate the EDID in /sys with a /dev/i2c device is to read the EDID from the /dev/i2c device

Also, please execute **ddcutil detect --verblose --f2** and submit the output.  This probes how /sys is used and may tell me more.

(Options ***--f2*** etc. are "utility" options for testing code in various ways.   Their meaning is not fixed.)

In any event, the problem is not in i2c-tools - that just provides the diagnostic utility i2cdetect. The related driver i2c-dev is also not the locus of the problem.  It is a unified way for userspace programs to access (almost) all the i2c devices which are implemented by a myriad of drivers.  In this case, i2c-dev just routes requests to the i915 video driver, which does the work.

Ragnoroct, 2025-02-13:
I don't have any DisplayPort-to-Thunderbolt cables to use, so I can't test that.
This is a lot of good info thanks.

[ddcutil-detect-f2.log](https://github.com/user-attachments/files/18786739/ddcutil-detect-f2.log)

rockowitz, 2025-02-13:
The ***--f2*** output is disappointing.

What I see is that connector card1-DP-5, as well as card1-DP-7 has  a valid EDID.  Unfortunately, sysfs doesn't provide the I2C bus numbers for these connectors, since they are on the dock.  Consequently, to associate the connector with an I2C bus requires matching the EDIDs.  But the EDID read from the bus is invalid, so matching won't work.