rockowitz/ddcutil #492: AOC PD32M - Odd issues.

vindoloop, 2025-02-07:
Well the first odd issue I'm having I discovered through the use of your software and your recommendation to try the product from enTech... in real time monitoring of the interface, it's discovering a frequent OSD button event.

[image]

12:42:06.69791...Event signaled: new VCP control value
12:42:06.70492...VCP table update
12:42:06.77593...Update VCP table - elapsed time = 78ms

Not sure what is happening here or what problems it might be causing. The monitor has an audible clicking noise that runs continuously through the speakers, I have to mute them to not go crazy, but that's ok cuz who uses built in monitor speakers anyways...

The second thing is, the usb key that came with my monitor has a firmware on it that is recent, but I can't correlate to what version the monitor currently has, and the g-menu software doesn't detect it, or offer any suggestions similar to what other people have posted on other forums.

Any guidance is appreciated, I figured the people here, would be the people to ask.

rockowitz, 2025-02-08:
I'm afraid I have nothing to suggest.  Perhaps someone else here will.  If your able to figure out more, please post what you've found so others can learn from it.

cnjsstong, 2025-09-20:
@vindoloop Hi! Have you figured out how to upgrade the firmware for PD32M? Mine can not wake up from sleep unless power off and on. Could you please share the firmware file as well?