rockowitz/ddcutil #493: Dell U4323QE with KVM - no displays found

rogerguess, 2025-02-09:
Dell U4323QE with KVM. 

**My problem is even with DDC/CI turned ON in the monitor settings, ddcutil is not able to communicate with the monitor.**

The end goal is a Raspberry Pi Zero 2 with 4 buttons for fast switching. I have two Macs, a win pc and a jetson orin nano. The Dell software for switching just does not cut it. I have tried from the RPI and from my Jetson running ubuntu. Neither can seem to see the monitor.

output from  detect, environment, interrogate:

<pre><font color="#26A269"><b>roger@orin-nano</b></font>:<font color="#12488B"><b>~</b></font>$ sudo ddcutil detect
[sudo] password for roger: 
(i2c_set_addr                  ) set_addr(/dev/i2c-0,I2C_SLAVE,0x50) failed, error = EBUSY
No displays found.
Is DDC/CI enabled in the monitor&apos;s on screen display?
Run &quot;ddcutil environment&quot; to check for system configuration problems.
<font color="#26A269"><b>roger@orin-nano</b></font>:<font color="#12488B"><b>~</b></font>$ ddcutil environment
The following tests probe the runtime environment using multiple overlapping methods.
*** Basic System Information ***

ddcutil version: 1.2.2

/proc/version:
   Linux version 5.15.148-tegra (buildbrain@mobile-u64-6344-d7000) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot 2022.08) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Thu Dec 12 15:37:09 PST 2024

Architecture:     aarch64
Distributor id:   Ubuntu
Release:          22.04
Found a known architecture

/proc/cmdline:
   root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 nospectre_bhb video=efifb:off console=tty0 bl_prof_dataptr=2031616@0x271E10000 bl_prof_ro_ptr=65536@0x271E00000 

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Current user: roger (1000)

Checking /dev/i2c-* devices...

Devices /dev/i2c-* must exist and the logged on user must have read/write permission for those devices (or at least those devices associated 
with monitors).

Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c

Alternatively, this can be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89,  0 Nov 21  2023 /dev/i2c-0
   crw-rw---- 1 root i2c 89,  1 Nov 21  2023 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 10 Nov 21  2023 /dev/i2c-10
   crw-rw---- 1 root i2c 89, 11 Feb  1 21:16 /dev/i2c-11
   crw-rw---- 1 root i2c 89,  2 Nov 21  2023 /dev/i2c-2
   crw-rw---- 1 root i2c 89,  4 Nov 21  2023 /dev/i2c-4
   crw-rw---- 1 root i2c 89,  5 Nov 21  2023 /dev/i2c-5
   crw-rw---- 1 root i2c 89,  7 Nov 21  2023 /dev/i2c-7
   crw-rw---- 1 root i2c 89,  9 Nov 21  2023 /dev/i2c-9

Current user (roger) has RW access to all /dev/i2c-* devices.

*** Primary Check 3: Check that module i2c_dev is loaded ***

Checking for driver i2c_dev...
   sysfs reports module i2c_dev is NOT loaded.
   Module i2c_dev is built into the kernel

*** Primary Check 4: Driver specific checks ***

Performing driver specific checks...
No driver specific checks apply.

*** Additional probes ***

Examining /sys/bus/i2c/devices...
   /sys/bus/i2c/devices/4-003c/name:  vrs-pseq
   /sys/bus/i2c/devices/i2c-0/name:   3160000.i2c
   /sys/bus/i2c/devices/i2c-1/name:   c240000.i2c
   /sys/bus/i2c/devices/i2c-2/name:   3180000.i2c
   /sys/bus/i2c/devices/i2c-4/name:   Tegra BPMP I2C adapter
   /sys/bus/i2c/devices/i2c-5/name:   31b0000.i2c
   /sys/bus/i2c/devices/i2c-7/name:   c250000.i2c
   /sys/bus/i2c/devices/i2c-9/name:   i2c-2-mux (chan_id 1)
   /sys/bus/i2c/devices/i2c-10/name:  i2c-2-mux (chan_id 0)
   /sys/bus/i2c/devices/10-0010/name: imx219
   /sys/bus/i2c/devices/9-0010/name:  imx219
   /sys/bus/i2c/devices/i2c-11/name:  NVIDIA SOC i2c adapter 0
   /sys/bus/i2c/devices/1-0025/name:  fusb301
   /sys/bus/i2c/devices/1-0040/name:  ina3221
   /sys/bus/i2c/devices/0-0050/name:  24c02
   /sys/bus/i2c/devices/0-0057/name:  24c02

Env_Accumulator:
   architecture:                  aarch64
   distributor_id                 Ubuntu
   Drivers detected:              
   /dev/i2c device numbers:       0 1 2 4 5 7 9 10 11
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 4 5 7 9 10 11
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        true
   loadable_i2c_dev_exists:       false
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     roger
   cur_uid:                       1000
   cur_user_in_group_i2c:         true
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.

<font color="#26A269"><b>roger@orin-nano</b></font>:<font color="#12488B"><b>~</b></font>$ 
</pre>

brtwrs, 2025-02-18:
Some Dell monitors have a wrong EDID header when using Higher Primary resolution. You can switch to a Lower Primary resolution. Somehow the header is now correct. 

https://www.dell.com/support/kbdoc/en-us/000182893/dell-u4021qw-monitor-has-abnormal-display-on-chrome-os

brtwrs, 2025-12-05:
I depends on the model i guess. But on the ones we are using I can press and hold the select button when selecting HDMI as input to switch between lower and higher primary resolution. On the lower setting the header is correct and the monitor is detected. Let me know if that works!