rockowitz/ddcutil #495: Unable to control brightness on 1 monitor

kaishi00, 2025-02-17:
[ddcutil output.txt](https://github.com/user-attachments/files/18818813/ddcutil.output.txt)

The output seems to be too long, but the relevant display that's not letting me adjust brightness is display 1 (/dev/i2c-7). The other 2 monitors both allow me to adjust brightness. Any ideas how to solve this?

rockowitz, 2025-02-18:
The output you've submitted contains only the final portion of the **interrogate** output.   Please submit the entire output.  Yes, it's voluminous because it's intended to cover a wide range of cases.

kaishi00, 2025-02-18:
> The output you've submitted contains only the final portion of the **interrogate** output. Please submit the entire output. Yes, it's voluminous because it's intended to cover a wide range of cases.

thank you, but the output was so long, I was only able to capture that much. Is there a way to capture the whole log?

rockowitz, 2025-02-19:
Redirect the output to a file: 
`ddcutil interrogate > whatever_you_choose_to_name_the_file`

kaishi00, 2025-02-19:
> Redirect the output to a file: `ddcutil interrogate > whatever_you_choose_to_name_the_file`

[ddcutil-output.txt](https://github.com/user-attachments/files/18868198/ddcutil-output.txt)

here is the complete output. I've noticed the issue is intermittent, or there's some kind of trigger. The controls aren't there (for only that 1 monitor, the 32M2V) after a reboot, but at some point as I use the computer, it'll randomly(?) show up again.

rockowitz, 2025-02-20:
Re the 32M2V on /dev/i2c-7, the communication error is DDCRC_DATA  the response data packet is garbled. The interrogate output attempt to read feature x10.  If fails twice with status DDCRC_DATA, but then succeeds on the 3rd try.  This is consistent with your observation that sometime it works, sometimes it doesn't.   The one mechanism that ddcutil has for coping with bad data, which often but not always works, is increase the time between sending a request packet and trying to read the reply packet.  Try using a high sleep-multiplier value, e.g. ***--sleep-multiplier 2.0***. 

Re the Samsung Odyssey, you say that you're able to adjust brightness.  I'm surprised.  The communication error reported is a hard EIO.  Consistent with that, the i2cdectect command does not report that slave address x37 (DDC) is responsive.

kaishi00, 2025-02-20:
> Re the 32M2V on /dev/i2c-7, the communication error is DDCRC_DATA the response data packet is garbled. The interrogate output attempt to read feature x10. If fails twice with status DDCRC_DATA, but [...]

Sorry I'm fairly new to linux, I tried running ddcutil -sleep-multiplier 2.0 and get a "No command specified" error. Which command do I need to run with that option? detect?

rockowitz, 2025-02-20:
***--sleep-multiplier*** is an option.  It modifies the behavior of any command that executes DDC commuminication with the monitor, e.g. **detect**, **getvcp**, **setvcp**.