rockowitz/ddcutil #505: External monitor detected in displays; black screen/"no signal" - Dell WD19TB dock / ThinkPad X1 Carbon Gen 10 / Dell

bjmc, 2025-02-27:
System info:

> **Laptop:** 21CBCTO1WW (LENOVO_MT_21CB_BU_Think_FM_ThinkPad X1 Carbon Gen 10)
> **OS:** Ubuntu 24.04 / kernel 6.8.0-52-generic
> **Dock:** Dell WD19TB Thunderbolt Dock
> **Monitor:** Dell S2721QS connected via HDMI (to the dock)
> **ddcutil version:** 1.4.1

This same monitor works with this same laptop when connected directly via HDMI (bypassing the dock). Also, this same monitor works with this same Thunderbolt dock, when connected to a different computer.

```
$  ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-11
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               BOE - BOE
      Model:                
      Product code:         2761  (0x0ac9)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2021,  Week: 40
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:  /dev/i2c-14
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL S2721QS
      Product code:         41366  (0xa196)
      Serial number:        BJB1M43
      Binary serial number: 810047052 (0x3048564c)
      Manufacture year:     2021,  Week: 12
   DDC communication failed

```
ddcutil interrogate output:
[interrogate.log](https://github.com/user-attachments/files/19009180/interrogate.log)

rockowitz, 2025-03-01:
First, you're using an ancient version of ddcutil. Release 2.2.0 is available here.

That said, given that you have a blank screen, the docking station and your laptop clearly have compatibility issues beyond DDC.  If you search here for Dell WD19TB you'll see that there are a number of reports of problems with this docking station. You mention that the monitor and dock work with another computer.  If that's a Windows computer it suggests that the i915 driver can't handle this configuration, which is not surprising.

rockowitz, 2025-03-06:
I just acquired a similar Dell WD17S dock.  What I saw on both Windows and Linux was that the HDMI screen was blank when connected to the dock.  On Windows, rebooting solved the problem.  On Linux, the monitor had to be explicitly enabled in the display configuration settings. ddcutil works fine.