rockowitz/ddcutil #506: 2.2.0 fails to build with --disable-drm

kostadinsh, 2025-02-27:
ddcutil 2.2.0 fails to build with --disable-drm. First bad commit is https://github.com/rockowitz/ddcutil/commit/aeb0b58

```
ddcutil 2.2.0
        version suffix:   
        =============

        libtool version         7:0:2

        prefix:                 /usr/local
        exec_prefix:            ${prefix}
        libexecdir:             ${exec_prefix}/libexec
        bindir:                 ${exec_prefix}/bin
        libdir:                 ${exec_prefix}/lib
        datarootdir:            ${prefix}/share
        datadir:                ${datarootdir}
        docdir:                 ${datarootdir}/doc/${PACKAGE_TARNAME}
        mandir:                 ${datarootdir}/man
        includedir:             ${prefix}/include
        pkgconfigdir:           
        required_packages:       glib-2.0 >= 2.40 jansson >= 2.0

        enable_lib:             yes
        enable_install_lib_only no
        enable_build_timestamp  yes
        enable_envcmds          yes
        enable_udev             yes
        enable_usb:             yes
        enable_drm:             no
        enable_x11:             yes
        enable_asan:            no
        enable_static_functions_visible:  no

        Developer-only options:
        enable_targetbsd:       no
        enable_doxygen:         no
        enable_failsim:         no
        include_testcases:      no

        compiler:               gcc
        CFLAGS:                 -g -O2
        CPPFLAGS:               
        LDFLAGS:
```

```
In file included from drm_connector_state.h:13,
                 from base_services.c:15:
/usr/include/xf86drmMode.h:43:10: fatal error: drm.h: No such file or directory
   43 | #include <drm.h>
      |          ^~~~~~~
compilation terminated.
In file included from drm_connector_state.c:18:
/usr/include/xf86drm.h:40:10: fatal error: drm.h: No such file or directory
   40 | #include <drm.h>
      |          ^~~~~~~
```

rockowitz, 2025-02-28:
Can I ask what sort of system you have without drm?  While the fix for --disable-x11 was straightforward, reimplementing --disable-drm is not.

rockowitz, 2025-03-01:
configure option ***--disable-drm*** has been reimplemented.  If built with configure option ***--disable-drm***, ddcutil does not make use of DRM APIs.  This does not block reading of /sys, the contents of which can be set by DRM.  I don't have a true drm-less system to test against, it's possible that there are unhandled errors in that situation.

Also added is configure  option ***--disable-watch-displays***, controlling whether libddcutil watches for display connection changes.  ***--disable-drm*** forces ***--disable-watch-displays***.

kostadinsh, 2025-08-01:
2.2.1 builds fine, but when running the test suite with `make check` I am getting:
`/usr/lib/gcc/x86_64-pc-linux-gnu/16/../../../../x86_64-pc-linux-gnu/bin/ld: ../.libs/libddcutil.so: undefined reference to 'dw_redetect_displays'`

rockowitz, 2025-08-02:
I assume that by "when running the test suite" you meant building with "make check" and did not mean executing the demo programs created by make check. 

Commit 2ed9275c in branch 2.2.1-dev allows building to complete without the undefined reference error when configure option --disable-drm was specified.  API function ddca_redetect_displays() returns DDCA_INVALID_OPERATION in this case.  (An alternative code path through ddca_redetect_displays() is possible for the --disable-drm case, but is non-trivial and so not implemented for now.)