rockowitz/ddcutil #507: Failed to find connector name for /dev/i2c-0 using EDID

gadefox, 2025-03-02:
I'm suddenly getting weird error message. I'm using `Debian trixie`.
Any idea?

`ddcutil --verbose getvcp 10`
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-0 using EDID 0x55ac48c67544
Failed to find connector name for /dev/i2c-0, set_connector_for_businfo_using_edid at line 1236 in file i2c_bus_core.c.
   I2C_Bus_Info at: 0x55ac48c78730
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-0 found:   true
   Bus /dev/i2c-0 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   last_checked_asleep:       false
   Display connectors reported by /sys:
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-2 using EDID 0x55ac48c79f04
Failed to find connector name for /dev/i2c-2, set_connector_for_businfo_using_edid at line 1236 in file i2c_bus_core.c.
   I2C_Bus_Info at: 0x55ac48c67ca0
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-2 found:   true
   Bus /dev/i2c-2 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   last_checked_asleep:       false
   Display connectors reported by /sys:
Display not found

`sudo ddcutil interrogate`
` ddcutil usbenvironment --verbose`
https://limewire.com/?referrer=pq7i8xx7p2

rockowitz, 2025-03-02:
Debian trixie now uses recently released ddcutil 2.2.0.   Your interrogate output reports release 2.2.0-dev.  As first step, please confirm that the problem exists with branch 2.2.0-release or 2.2.1-dev, and interrogate.  I suspect that the problem relates to the fact that older video drivers do not populate the connector_id attribute in /sys/class/drm/cardN-xxx and the recovery algorithm is not working.

gadefox, 2025-03-02:
I have no idea why it reports `dev`, because it's the official package. I reinstalled the tool again (`apt purge; apt install`) and `ddcutil --version` still shows 2.2.0-dev. The `ddcutil` has not been updated. The package manager `apt` updated `i2c-tools (4.4-1+b3)` and `libi2c0:amd64 (4.4-1+b3)` on Thursday.

gadefox, 2025-03-02:
I just built the version 2.1.0, which works fine..

`cd ~/.build/ddcutil/src`
`./ddcutil --version`
ddcutil 2.1.0
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

`./ddcutil getvcp 10`
VCP code 0x10 (Brightness                    ): current value =    20, max value =   100

gadefox, 2025-03-03:
My solution for now is a downgrade to previous stable version (2.1.4) which i built from the source.

`ddcutil --version`
ddcutil 2.1.4
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

`ddcutil getvcp 10`
VCP code 0x10 (Brightness                    ): current value =    20, max value =   100

Quick question:
The `configure` script checks if the `drm` library (include and `.pc` file) is available. Should I build the `ddcutil` tool with `drm`? Thanks..

rockowitz, 2025-03-06:
The messages you started seeing should have gone only to the system log, not to the terminal. In your situation they are informational only, reflecting the fact that the Nvidia driver on your system does not, apparently, implement DRM.  That  is only a problem when the shared library is checking for display connection and disconnection.

I've made changes in branch 2.2.1-dev that should address this situation.

rockowitz, 2025-04-28:
If you build from branch 2.2.1-dev does the problem persist?

gadefox, 2025-04-28:
Yes, see bellow.
```
~/.build/ddcutil-2.2.1-dev
❯ ./src/ddcutil getvcp 10
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-0 using EDID 0x55987ca1f9b4
Failed to find connector name for /dev/i2c-0, set_connector_for_businfo_using_edid at line 1345 in file i2c_bus_core.c. 
   I2C_Bus_Info at: 0x55987ca36050
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-0 found:   true
   Bus /dev/i2c-0 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   Display connectors reported by /sys:
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-2 using EDID 0x55987ca25614
```

Build options:
```
	ddcutil 2.2.1
	version suffix:   dev
	=============

	libtool version  	7:0:2

	prefix:                 /usr/local
	exec_prefix:            ${prefix}
	libexecdir:             ${exec_prefix}/libexec
	bindir:                 ${exec_prefix}/bin
	libdir:                 ${exec_prefix}/lib
	datarootdir:            ${prefix}/share
	datadir:                ${datarootdir}
	docdir:                 ${datarootdir}/doc/${PACKAGE_TARNAME}
	mandir:               	${datarootdir}/man
	includedir:             ${prefix}/include
	pkgconfigdir:           
	required_packages:       glib-2.0 >= 2.40 jansson >= 2.0

	enable_lib:             yes
	enable_install_lib_only no
	enable_build_timestamp  yes
	enable_envcmds          yes
	enable_udev             yes
	enable_usb:             yes
	enable_drm:             no
	enable_watch_displays:  no
	enable_x11:             yes
	enable_asan:            no
	enable_static_functions_visible:  no

	Developer-only options:
	enable_targetbsd:       no
	enable_doxygen:         no
	enable_failsim:         no
	include_testcases:      no

	compiler:               gcc
	CFLAGS:                 -g -O2
	CPPFLAGS:               
	LDFLAGS:                
```

gadefox, 2025-05-08:
@rockowitz you can add and logs and I'm happy to build temporary source code and make few tests if you want..

rockowitz, 2025-05-23:
Re the getvcp 10 output: Was that the extent of output on the command?  The diagnostic output should have gone to the system log, not the terminal.  Did you build from the current 2.2.1-dev branch?

Re the ddcutil-dev-interrogate output, were you running as root?  

Please execute the following commands and submit the output: 

**ls -R /sys/class/drm/* **

**ddcutil detect --trcfunc get_i2c_driver_info**

**ddcutil detect --f23**

(***--f23*** is a development option that forces ddcutil to regard the driver as "primitive", i.e. /sys/class/drm is not fully populated)

gadefox, 2025-05-23:
ddcutil getvcp 10
```
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-0 using EDID 0x55ee9a3846a4
Failed to find connector name for /dev/i2c-0, set_connector_for_businfo_using_edid at line 1345 in file i2c_bus_core.c. 
   I2C_Bus_Info at: 0x55ee9a398ea0
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-0 found:   true
   Bus /dev/i2c-0 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   Display connectors reported by /sys:
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-2 using EDID 0x55ee9a3987c4
```

ddcutil detect --f23
```
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-0 using EDID 0x5582850e7064
Failed to find connector name for /dev/i2c-0, set_connector_for_businfo_using_edid at line 1345 in file i2c_bus_core.c. 
   I2C_Bus_Info at: 0x5582850e6360
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-0 found:   true
   Bus /dev/i2c-0 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   Display connectors reported by /sys:
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-2 using EDID 0x5582850e91a4
```

ddcutil detect --trcfunc get_i2c_driver_info
```
(get_i2c_driver_info           ) Starting  busno=0, depth=-1
      /sys/bus/i2c/devices/i2c-0/name                                       =  NVIDIA i2c adapter 0 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55800c1c8760
   Sysfs_I2C_Info at: 0x55800c1c8760
      busno:                     0
      name:                      NVIDIA i2c adapter 0 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
      adapter_class:             0x030000
      driver:                    nvidia
      driver_version:            340.108
      conflicting_driver_names:  
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-0 using EDID 0x55800c1e0e34
Failed to find connector name for /dev/i2c-0, set_connector_for_businfo_using_edid at line 1345 in file i2c_bus_core.c. 
   I2C_Bus_Info at: 0x55800c1cb370
   Flags:                   I2C_BUS_EXISTS | I2C_BUS_ACCESSIBLE | I2C_BUS_PROBED | I2C_BUS_X50_EDID
   Bus /dev/i2c-0 found:   true
   Bus /dev/i2c-0 probed:  true
   errno for open:          OK(0): success
   drm_connector_found_by:  DRM_CONNECTOR_NOT_CHECKED (0)
   Display connectors reported by /sys:
(get_i2c_driver_info           ) Starting  busno=1, depth=-1
      /sys/bus/i2c/devices/i2c-1/name                                       =  NVIDIA i2c adapter 1 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55800c1cf230
   Sysfs_I2C_Info at: 0x55800c1cf230
      busno:                     1
      name:                      NVIDIA i2c adapter 1 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
      adapter_class:             0x030000
      driver:                    nvidia
      driver_version:            340.108
      conflicting_driver_names:  
(get_i2c_driver_info           ) Starting  busno=2, depth=-1
      /sys/bus/i2c/devices/i2c-2/name                                       =  NVIDIA i2c adapter 2 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55800c1cf230
   Sysfs_I2C_Info at: 0x55800c1cf230
      busno:                     2
      name:                      NVIDIA i2c adapter 2 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
      adapter_class:             0x030000
      driver:                    nvidia
      driver_version:            340.108
      conflicting_driver_names:  
(set_connector_for_businfo_using_edid)           Failed to find connector name for /dev/i2c-2 using EDID 0x55800c1cf344
```

ls -R /sys/class/drm/*
```
/sys/class/drm/version

/sys/class/drm/card0:
dev  device  power  subsystem  uevent

/sys/class/drm/card0/power:
async  autosuspend_delay_ms  control  runtime_active_kids  runtime_active_time  runtime_enabled  runtime_status  runtime_suspended_time  runtime_usage
```

sudo ddcutil interrogate > ~/ddcutil.interrogate

[ddcutil.interrogate.gz](https://github.com/user-attachments/files/20416093/ddcutil.interrogate.gz)

rockowitz, 2025-05-24:
I've updated 2.2.1-dev to more correctly detect that /sys/class/drm is not populated so various optimizations are not used.  You should no longer be seeing all the diagnostic messages, and output from **detect**, **getvcp**, etc. should appear as normal.  Output from **interrogate** should now also be normal, but I'm concerned that some issues may still exist.  Let me know how it looks and submit any output that looks suspicious. Thanks.

gadefox, 2025-05-24:
Yep, we're closer. Output from interrougate seems to be good. Please check this out..

ddcutil getvcp 10
```
Display not found
```
ddcutil detect --f23
```
Invalid display
   I2C bus:  /dev/i2c-0
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQG2222HDL
      Product code:         30809  (0x7859)
      Serial number:        PBB51606067
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2011,  Week: 47
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)

Invalid display
   I2C bus:  /dev/i2c-2
   EDID synopsis:
      Mfg id:               SEC - Seiko Epson Corporation
      Model:                
      Product code:         21069  (0x524d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2009,  Week: 0
   This appears to be a laptop display.  Laptop displays do not support DDC/CI.
```
ddcutil detect --trcfunc get_i2c_driver_info
```
(get_i2c_driver_info           ) Starting  busno=0, depth=-1
      /sys/bus/i2c/devices/i2c-0/name                                       =  NVIDIA i2c adapter 0 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55ec986be900
   Sysfs_I2C_Info at: 0x55ec986be900
      busno:                     0
      name:                      NVIDIA i2c adapter 0 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
      adapter_class:             0x030000
      driver:                    nvidia
      driver_version:            340.108
      conflicting_driver_names:  
(get_i2c_driver_info           ) Starting  busno=1, depth=-1
      /sys/bus/i2c/devices/i2c-1/name                                       =  NVIDIA i2c adapter 1 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55ec986a00f0
   Sysfs_I2C_Info at: 0x55ec986a00f0
      busno:                     1
      name:                      NVIDIA i2c adapter 1 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
      adapter_class:             0x030000
      driver:                    nvidia
      driver_version:            340.108
      conflicting_driver_names:  
(get_i2c_driver_info           ) Starting  busno=2, depth=-1
      /sys/bus/i2c/devices/i2c-2/name                                       =  NVIDIA i2c adapter 2 at 1:00.0
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/class               =  0x030000
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver              -> nvidia
      /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/driver/module/version=  340.108
(get_i2c_driver_info           ) Returning "Sysfs_I2C_Info" at 0x55ec986a00f0
   Sysfs_I2C_Info at: 0x55ec986a00f0
      busno:                     2
      name:                      NVIDIA i2c adapter 2 at 1:00.0
      adapter_path:              /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0
[... 32 lines cut ...]
Invalid display
   I2C bus:  /dev/i2c-0
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQG2222HDL
      Product code:         30809  (0x7859)
      Serial number:        PBB51606067
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2011,  Week: 47
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)

Invalid display
   I2C bus:  /dev/i2c-2
   EDID synopsis:
      Mfg id:               SEC - Seiko Epson Corporation
      Model:                
      Product code:         21069  (0x524d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2009,  Week: 0
   This appears to be a laptop display.  Laptop displays do not support DDC/CI.
```
sudo ddcutil interrogate > ~/ddcutil.interrogate

[ddcutil.interrogate.gz](https://github.com/user-attachments/files/20422186/ddcutil.interrogate.gz)

rockowitz, 2025-05-24:
With the latest 2.2.1-dev, please execute the following and submit the output: 

**ddcutil detect --trcfrom i2c_detect_x37**

**ddcutil detect --trcfrom i2c_detect_x37 --use-file-io**

I suspect you're hitting an old nvidia driver issue where use of i2c-dev's ioctl interface fails.

gadefox, 2025-05-24:
ddcutil detect --trcfrom i2c_detect_x37
```
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-0, driver=nvidia
(i2c_ioctl_writer              ) Starting  fh=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, pbytes=0x7ffd9974fa9b -> 00
(i2c_ioctl_writer              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0
(i2c_detect_x37                )           invoke_ioctl_writer() for slave address x37 returned EINVAL(-22)
(i2c_ioctl_reader              ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, read_bytewise=false, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             )           Error in ioctl() read, rc=-1, errno=EINVAL(-22): Invalid argument, device=/dev/i2c-0
(i2c_ioctl_reader1             ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_ioctl_reader              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_detect_x37                )           invoke_ioctl_reader() for slave address x37 returned EINVAL(-22)
(i2c_detect_x37                )           driver=nvidia, sleeping for 2000 millisec
(i2c_ioctl_writer              ) Starting  fh=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, pbytes=0x7ffd9974fa9b -> 00
(i2c_ioctl_writer              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0
(i2c_detect_x37                )           invoke_ioctl_writer() for slave address x37 returned EINVAL(-22)
(i2c_ioctl_reader              ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, read_bytewise=false, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             )           Error in ioctl() read, rc=-1, errno=EINVAL(-22): Invalid argument, device=/dev/i2c-0
(i2c_ioctl_reader1             ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_ioctl_reader              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_detect_x37                )           invoke_ioctl_reader() for slave address x37 returned EINVAL(-22)
(i2c_detect_x37                )           driver=nvidia, sleeping for 2000 millisec
(i2c_ioctl_writer              ) Starting  fh=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, pbytes=0x7ffd9974fa9b -> 00
(i2c_ioctl_writer              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0
(i2c_detect_x37                )           invoke_ioctl_writer() for slave address x37 returned EINVAL(-22)
(i2c_ioctl_reader              ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, read_bytewise=false, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, bytect=4, readbuf=0x7ffd9974fa9c
(i2c_ioctl_reader1             )           Error in ioctl() read, rc=-1, errno=EINVAL(-22): Invalid argument, device=/dev/i2c-0
(i2c_ioctl_reader1             ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_ioctl_reader              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_detect_x37                )           invoke_ioctl_reader() for slave address x37 returned EINVAL(-22)
(i2c_detect_x37                )           driver=nvidia, sleeping for 2000 millisec
(i2c_detect_x37                ) Done      Returning: EINVAL(-22). loopctr=3
Invalid display
   I2C bus:  /dev/i2c-0
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQG2222HDL
      Product code:         30809  (0x7859)
      Serial number:        PBB51606067
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2011,  Week: 47
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)

Invalid display
   I2C bus:  /dev/i2c-2
   EDID synopsis:
      Mfg id:               SEC - Seiko Epson Corporation
      Model:                
      Product code:         21069  (0x524d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2009,  Week: 0
   This appears to be a laptop display.  Laptop displays do not support DDC/CI.
```

ddcutil detect --trcfrom i2c_detect_x37 --use-file-io
```
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-0, driver=nvidia
(i2c_ioctl_writer              ) Starting  fh=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, pbytes=0x7fff7a441dab -> 00
(i2c_ioctl_writer              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0
(i2c_detect_x37                )           invoke_ioctl_writer() for slave address x37 returned EINVAL(-22)
(i2c_ioctl_reader              ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, read_bytewise=false, bytect=4, readbuf=0x7fff7a441dac
(i2c_ioctl_reader1             ) Starting  fd=3, fn=/dev/i2c-0, slave_addr=0x37, bytect=4, readbuf=0x7fff7a441dac
(i2c_ioctl_reader1             )           Error in ioctl() read, rc=-1, errno=EINVAL(-22): Invalid argument, device=/dev/i2c-0
(i2c_ioctl_reader1             ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
(i2c_ioctl_reader              ) Done      Returning: EINVAL(-22). fh=3, filename=/dev/i2c-0, readbuf: 00 00 00 00
[... cut ...]
      Manufacture year:     2011,  Week: 47
   This monitor does not support DDC/CI. (I2C slave address x37 is unresponsive.)

Invalid display
   I2C bus:  /dev/i2c-2
   EDID synopsis:
      Mfg id:               SEC - Seiko Epson Corporation
      Model:                
      Product code:         21069  (0x524d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2009,  Week: 0
   This appears to be a laptop display.  Laptop displays do not support DDC/CI.
```

gadefox, 2025-05-24:
ddcutil --version
```
ddcutil 2.1.4
Copyright (C) 2015-2024 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

ddcutil detect --trcfrom i2c_detect_x37
```
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-0
(invoke_i2c_writer             ) Starting  fd=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, bytes_to_write=0x7ffc419bc73b -> 00
(invoke_i2c_writer             )           strategy = I2C_IO_STRATEGY_FILEIO
(i2c_fileio_writer             ) Starting  fd=3, filename=/dev/i2c-0, slave_address=0x37, bytect=1, pbytes=0x7ffc419bc73b -> 00
(i2c_set_addr                  ) Starting  fd=3, addr=0x37, filename=/dev/i2c-0, i2c_forceable_slave_addr_flag=false
(i2c_set_addr0                 ) Starting  fd=3, addr=0x37, filename=/dev/i2c-0, op=I2C_SLAVE
(i2c_set_addr0                 ) Done      Returning: OK(0). 
(i2c_set_addr                  ) Done      Returning: OK(0). 
(i2c_fileio_writer             ) Done      Returning: OK(0). 
(invoke_i2c_writer             ) Done      Returning: OK(0). 
(i2c_detect_x37                )           invoke_i2c_writer() for slave address x37 returned OK(0)
(i2c_detect_x37                ) Done      Returning: OK(0). 
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-1
(invoke_i2c_writer             ) Starting  fd=3, filename=/dev/i2c-1, slave_address=0x37, bytect=1, bytes_to_write=0x7ffc419bc73b -> 00
(invoke_i2c_writer             )           strategy = I2C_IO_STRATEGY_FILEIO
(i2c_fileio_writer             ) Starting  fd=3, filename=/dev/i2c-1, slave_address=0x37, bytect=1, pbytes=0x7ffc419bc73b -> 00
(i2c_set_addr                  ) Starting  fd=3, addr=0x37, filename=/dev/i2c-1, i2c_forceable_slave_addr_flag=false
(i2c_set_addr0                 ) Starting  fd=3, addr=0x37, filename=/dev/i2c-1, op=I2C_SLAVE
(i2c_set_addr0                 ) Done      Returning: OK(0). 
(i2c_set_addr                  ) Done      Returning: OK(0). 
(i2c_fileio_writer             ) Done      Returning: EIO(-5). 
(invoke_i2c_writer             ) Done      Returning: EIO(-5). 
(i2c_detect_x37                )           invoke_i2c_writer() for slave address x37 returned EIO(-5)
(invoke_i2c_reader             ) Starting  fd=3, filename=/dev/i2c-1, slave_address=0x37, bytect=4, read_bytewise=false, readbuf=0x7ffc419bc73c
(invoke_i2c_reader             )           strategy = I2C_IO_STRATEGY_FILEIO
(i2c_fileio_reader             ) Starting  fd=3, fn=/dev/i2c-1, bytect=4, slave_address=0x37, single_byte_reads=false
(i2c_set_addr                  ) Starting  fd=3, addr=0x37, filename=/dev/i2c-1, i2c_forceable_slave_addr_flag=false
(i2c_set_addr0                 ) Starting  fd=3, addr=0x37, filename=/dev/i2c-1, op=I2C_SLAVE
(i2c_set_addr0                 ) Done      Returning: OK(0). 
(i2c_set_addr                  ) Done      Returning: OK(0). 
(i2c_fileio_reader             ) Done      Returning: EIO(-5). readbuf: 92 55 00 00
(invoke_i2c_reader             ) Done      Returning: EIO(-5). 
(i2c_detect_x37                )           invoke_i2c_reader() for slave address x37 returned EIO(-5)
(i2c_detect_x37                ) Done      Returning: EIO(-5). 
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-2
(invoke_i2c_writer             ) Starting  fd=3, filename=/dev/i2c-2, slave_address=0x37, bytect=1, bytes_to_write=0x7ffc419bc73b -> 00
(invoke_i2c_writer             )           strategy = I2C_IO_STRATEGY_FILEIO
(i2c_fileio_writer             ) Starting  fd=3, filename=/dev/i2c-2, slave_address=0x37, bytect=1, pbytes=0x7ffc419bc73b -> 00
(i2c_set_addr                  ) Starting  fd=3, addr=0x37, filename=/dev/i2c-2, i2c_forceable_slave_addr_flag=false
(i2c_set_addr0                 ) Starting  fd=3, addr=0x37, filename=/dev/i2c-2, op=I2C_SLAVE
(i2c_set_addr0                 ) Done      Returning: OK(0). 
(i2c_set_addr                  ) Done      Returning: OK(0). 
(i2c_fileio_writer             ) Done      Returning: EIO(-5). 
(invoke_i2c_writer             ) Done      Returning: EIO(-5). 
(i2c_detect_x37                )           invoke_i2c_writer() for slave address x37 returned EIO(-5)
(invoke_i2c_reader             ) Starting  fd=3, filename=/dev/i2c-2, slave_address=0x37, bytect=4, read_bytewise=false, readbuf=0x7ffc419bc73c
(invoke_i2c_reader             )           strategy = I2C_IO_STRATEGY_FILEIO
(i2c_fileio_reader             ) Starting  fd=3, fn=/dev/i2c-2, bytect=4, slave_address=0x37, single_byte_reads=false
(i2c_set_addr                  ) Starting  fd=3, addr=0x37, filename=/dev/i2c-2, i2c_forceable_slave_addr_flag=false
(i2c_set_addr0                 ) Starting  fd=3, addr=0x37, filename=/dev/i2c-2, op=I2C_SLAVE
(i2c_set_addr0                 ) Done      Returning: OK(0). 
(i2c_set_addr                  ) Done      Returning: OK(0). 
(i2c_fileio_reader             ) Done      Returning: EIO(-5). readbuf: 92 55 00 00
(invoke_i2c_reader             ) Done      Returning: EIO(-5). 
(i2c_detect_x37                )           invoke_i2c_reader() for slave address x37 returned EIO(-5)
(i2c_detect_x37                ) Done      Returning: EIO(-5). 
(i2c_detect_x37                ) Starting  fd=3 - /dev/i2c-3
(invoke_i2c_writer             ) Starting  fd=3, filename=/dev/i2c-3, slave_address=0x37, bytect=1, bytes_to_write=0x7ffc419bc73b -> 00
[... cut ...]
Invalid display
   I2C bus:  /dev/i2c-2
   EDID synopsis:
      Mfg id:               SEC - Seiko Epson Corporation
      Model:                
      Product code:         21069  (0x524d)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2009,  Week: 0
   DDC communication failed. (getvcp of feature x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO])
   This appears to be a laptop display.  Laptop displays do not support DDC/CI
```

rockowitz, 2025-05-24:
Please pull branch 2.2.1-dev and run the commands again.   There should be calls to invoke_i2c_reader() and invoke_i2c_writer() in the 2.2.1-dev output. Thanks.

gadefox, 2025-05-24:
Thanks a lot!

rockowitz, 2025-05-24:
Given the output from **ddcutil detect --trcfrom i2c_detect_x37**, you should see the following line in your system log (**journalctl -t ddcutil**): 
~~~
nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
~~~

Please confirm that's the case.  What's going on is that by default ddcutil uses the ioctl interface to i2c-dev.  If it encounters the nvidia/i2c-dev bug, it switches to using i2c-dev's file interface. That recovery has already occurred by the time i2c_detect_x37() is called.

Can you send me the current **interrogate** output?  I want to confirm that it looks ok as well.  Thanks.

gadefox, 2025-05-24:
journalctl -t ddcutil
```
-- Boot d419415573b845a1984c65921e2e7a98 --
May 24 19:34:46 hell ddcutil[1806]: [  1806] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 19:34:52 hell ddcutil[1806]: [  1806] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 19:34:52 hell ddcutil[1806]: [  1806] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 19:36:57 hell ddcutil[2158]: [  2158] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 19:36:57 hell ddcutil[2158]: [  2158] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 19:37:44 hell ddcutil[2191]: nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 19:37:46 hell ddcutil[2191]: busno=2, sleep-multiplier =  1.60. Testing for supported feature 0x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO]
May 24 19:38:49 hell ddcutil[2226]: busno=2, sleep-multiplier =  2.00. Testing for supported feature 0x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO]
May 24 20:32:27 hell ddcutil[21637]: [ 21637] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:32:27 hell ddcutil[21637]: [ 21637] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:32:27 hell ddcutil[21637]: [ 21637] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:33:28 hell ddcutil[21666]: [ 21666] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:33:28 hell ddcutil[21666]: [ 21666] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:34:33 hell ddcutil[21698]: [ 21698] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:34:33 hell ddcutil[21698]: [ 21698] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:34:33 hell ddcutil[21698]: [ 21698] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:34:55 hell ddcutil[21712]: [ 21712] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:34:55 hell ddcutil[21712]: [ 21712] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:34:55 hell ddcutil[21712]: [ 21712] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:35:11 hell ddcutil[21726]: [ 21726] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:35:11 hell ddcutil[21726]: [ 21726] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:35:11 hell ddcutil[21726]: [ 21726] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:37:52 hell ddcutil[21868]: [ 21868] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:37:52 hell ddcutil[21868]: [ 21868] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:37:52 hell ddcutil[21868]: [ 21868] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 20:37:55 hell ddcutil[21879]: nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 20:37:57 hell ddcutil[21879]: busno=2, sleep-multiplier =  2.00. Testing for supported feature 0x10 returned Error_Info[EIO in ddc_write_read_with_retry, causes: EIO]
May 24 21:44:42 hell ddcutil[25364]: [ 25364] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 21:44:42 hell ddcutil[25364]: [ 25364] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 21:44:42 hell ddcutil[25364]: [ 25364] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 21:44:46 hell ddcutil[25373]: [ 25373] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 21:44:47 hell ddcutil[25373]: [ 25373] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 21:44:47 hell ddcutil[25373]: [ 25373] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 21:44:48 hell ddcutil[25379]: [ 25379] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 21:44:49 hell ddcutil[25379]: [ 25379] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
[... cut ...]
May 24 21:45:05 hell ddcutil[25426]: [ 25426] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 22:30:31 hell ddcutil[26327]: [ 26327] nvida/i2c-dev bug encountered. Forcing future io to I2C_IO_STRATEGY_FILEIO. Retrying
May 24 22:30:31 hell ddcutil[26327]: [ 26327] is_sysfs_reliable_for_busno(0) returned false and session_type = X11. Using X11 to determine if display is asleep
May 24 22:30:31 hell ddcutil[26327]: [ 26327] is_sysfs_reliable_for_busno(2) returned false and session_type = X11. Using X11 to determine if display is asleep
```

gadefox, 2025-05-24:
ddcutil --version
```
ddcutil 2.2.1-dev
Copyright (C) 2015-2025 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

[ddcutil.interrogate.gz](https://github.com/user-attachments/files/20428053/ddcutil.interrogate.gz)

rockowitz, 2025-05-25:
Thank you for all your help on this!