rockowitz/ddcutil #508: External monitor not detected on laptop

PowerMan2206, 2025-03-10:
So I have a Lenovo Legion 5 laptop (pretty decent Linux support, honestly). I installed i2c and added the module and added myself to the `i2c` group and all that jazz. Oh and also I'm using nouveau drivers (although you can probably see that in the attachment).

Issue is, neither i2c nor ddcutil can detect my external Dell S2316H monitor. I got it connected to the one and only HDMI port on the laptop. I was thinking that maybe the issue is that the port goes to the dedicated Nvidia graphics card, but it seems to be detected with nothing found on it.

The monitor has a DDC/CI setting and I made sure it's enabled, but still no luck.

Got the output of `sudo ddcutil interrogate --verbose` [here.](https://github.com/user-attachments/files/19171953/ddcutil-output.txt)

rockowitz, 2025-03-12:
The interrogate output is very strange. The EDID reported for /dev/i2c-8 is sometimes that of the laptop display, not the HDMI display.  Try using option ***--disable-try-get-edid-from-sysfs***.

PowerMan2206, 2025-03-12:
Tried using that with `detect` with no luck. Tried running the `interrogate` one again, output seems a little different according to `diff`. [ddcutil-output2.txt](https://github.com/user-attachments/files/19213354/ddcutil-output2.txt)