rockowitz/ddcutil #509: ddcutil crashdump.

gjcarter2, 2025-03-11:
Mar 11 11:21:07 praxus.aesgi.com systemd-coredump[5973]: [🡕] Process 5967 (ddcutil) of user 0 dumped core.
                                                         
                                                         Module libcap.so.2 from rpm libcap-2.70-4.fc41.x86_64
                                                         Module libpcre2-8.so.0 from rpm pcre2-10.44-1.fc41.1.x86_64
                                                         Module libdrm.so.2 from rpm libdrm-2.4.124-1.fc41.x86_64
                                                         Module libudev.so.1 from rpm systemd-256.12-1.fc41.x86_64
                                                         Module libglib-2.0.so.0 from rpm glib2-2.82.2-1.fc41.x86_64
                                                         Module libjansson.so.4 from rpm jansson-2.13.1-10.fc41.x86_64
                                                         Module ddcutil from rpm ddcutil-2.1.2-2.fc41.x86_64
                                                         Stack trace of thread 5967:
                                                         #0  0x00007f9ffd494a9c __strlen_evex (libc.so.6 + 0x15aa9c)
                                                         #1  0x000056108c39d521 strsplit (ddcutil + 0x8d521)
                                                         #2  0x000056108c3adebc avoid_device_by_usb_interfaces_property_string (ddcutil + 0x9debc)
                                                         #3  0x000056108c3ae192 is_possible_monitor_by_hiddev_name (ddcutil + 0x9e192)
                                                         #4  0x000056108c3ae2c0 check_usb_monitor (ddcutil + 0x9e2c0)
                                                         #5  0x000056108c32d928 main (ddcutil + 0x1d928)
                                                         #6  0x00007f9ffd33d248 __libc_start_call_main (libc.so.6 + 0x3248)
                                                         #7  0x00007f9ffd33d30b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x330b)
                                                         #8  0x000056108c32da75 _start (ddcutil + 0x1da75)
                                                         ELF object binary architecture: AMD x86-64

This is on a Fedora 41 system, no external monitors and running rc6 of linux-6.14.
This also happens on the stock kernels as well.

Depending on the situation the crash can result in a locked up system, or just a crash dump.

This happens on my Alienware M18r1 AMD laptop with a M790.

gjcarter2, 2025-03-11:
Here is a dump of my PCI device list:
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:01.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Dummy Host Bridge
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Dummy Host Bridge
00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:03.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge GPP Bridge
00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Dummy Host Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Internal GPP Bridge to Bus [C:A]
00:08.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Internal GPP Bridge to Bus [C:A]
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 71)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge Data Fabric; Function 7
01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch (rev 10)
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch (rev 10)
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 31 [Radeon RX 7900 XT/7900 XTX/7900 GRE/7900M] (rev cf)
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Navi 31 HDMI/DP Audio
04:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller S4LV008[Pascal]
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
06:00.0 PCI bridge: ASMedia Technology Inc. Device 242a (rev 01)
07:02.0 PCI bridge: ASMedia Technology Inc. Device 242b (rev 01)
08:00.0 USB controller: ASMedia Technology Inc. Device 242c (rev 01)
09:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter
0a:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
0b:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0c:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge PCIe Dummy Function (rev d8)
0c:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 19h PSP/CCP
0c:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge USB 3.1 xHCI
0c:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge USB 3.1 xHCI
0c:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] ACP/ACP3X/ACP6x Audio Coprocessor (rev 62)
0c:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h/19h/1ah HD Audio Controller
0d:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Raphael/Granite Ridge USB 2.0 xHCI

rockowitz, 2025-03-12:
This looks like a very old bug that has been fixed.  However, it also looks like you're running an older version of ddcutil that sets permissions at startup for usb devices that might possibly be used for communication with the monitor's Virtual Control Panel.  Such monitors are rare, your error appears to occur when the related UDEV rule is executed at startup.  ddcutil 2.2.0 no longer automatically installs this udev rule.  Look for a file named like 60-ddcutil-usb.rules, most likely  in directory  /etc/udev/rules.d or /usr/lib/udev/rules.d, and delete it.

gjcarter2, 2025-03-12:
60-ddcutil-i2c.rules
and
60-ddcutil-usb.rules

I have deleted from /usr/lib/udev/rules.d

Not sure of the effect yet. Since the issue is random, I will report back next week.

gjcarter2, 2025-03-12:
ddcutil -V
ddcutil 2.1.2

Latest version seems to be 2.2.0.  FYI

gjcarter2, 2025-03-12:
Yes very old version 2.1.2.  Odd for Fedora as it usually tracks things
very closely.

Metal-spoon, 2025-03-12:
Getting the exact same coredumps, also on fedora.

```
Mar 12 10:10:47 Fedora-btw systemd-coredump[61481]: [🡕] Process 61469 (ddcutil) of user 0 dumped core.

Module libcap.so.2 from rpm libcap-2.70-4.fc41.x86_64
Module libpcre2-8.so.0 from rpm pcre2-10.44-1.fc41.1.x86_64
Module libdrm.so.2 from rpm libdrm-2.4.124-1.fc41.x86_64
Module libudev.so.1 from rpm systemd-256.12-1.fc41.x86_64
Module libglib-2.0.so.0 from rpm glib2-2.82.2-1.fc41.x86_64
Module libjansson.so.4 from rpm jansson-2.13.1-10.fc41.x86_64
Module ddcutil from rpm ddcutil-2.1.2-2.fc41.x86_64
Stack trace of thread 61469:
#0  0x00007feaa9e9481d __strlen_avx2 (libc.so.6 + 0x14a81d)
#1  0x00005615f5b77521 strsplit (ddcutil + 0x8d521)
#2  0x00005615f5b87ebc avoid_device_by_usb_interfaces_property_string (ddcutil + 0x9debc)
#3  0x00005615f5b88192 is_possible_monitor_by_hiddev_name (ddcutil + 0x9e192)
#4  0x00005615f5b882c0 check_usb_monitor (ddcutil + 0x9e2c0)
#5  0x00005615f5b07928 main (ddcutil + 0x1d928)
#6  0x00007feaa9d4d248 __libc_start_call_main (libc.so.6 + 0x3248)
#7  0x00007feaa9d4d30b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x330b)
#8  0x00005615f5b07a75 _start (ddcutil + 0x1da75)
ELF object binary architecture: AMD x86-64
```

Also on ddcutil 2.1.2.
Surely can't be a coincidence...

rockowitz, 2025-03-12:
You want to  delete 60-ddcutil-usb.rules but not 60-ddcutil-i2c.rules. The latter gives the logged on user RW access to /de/i2c devices associated with monitors without having to use group i2.

gjcarter2, 2025-03-12:
The crashes didn't stop though without deleting both of them.

gjcarter2, 2025-03-17:
So far so good.  No crashes, and with the recent fixes I got my Eclipse IDE isn't wigging out on the help menu anymore.

I made a note to the distro owners on Fedora about the ddcutil package being so old.  I didn't get a reply yet.

closed 2025-03-17 by gjcarter2