rockowitz/ddcutil #510: Building for LibreElec 12.0.2 via static linking

ChaosNicro, 2025-03-12:
Hey there,

I'd like to use ddcutil on my RPi4B running LibreElec 12.0.2. I have confirmed that the same setup works in RaspberryPi-OS. LE however does not provide libX11 or libjansson on RPi. I can disable X11 in configure easily enough when building in RPi-OS, though libjansson still makes it fail when ran in LE.

I did read #242 and tried static linking, however the ddcutil placed into the src-directory keeps relying on shared libraries with all options I've tried. So far I've tried  'autogen.sh --disable-x11 --disable-shared --enable-static' (X11 reports as disabled, LD_FLAGS is empty), and 'autogen.sh --disable-x11 --disable-shared --enable-static LDFLAGS=-static' (which does report -static in LD_FLAGS).

Is there any way I can package the shared libraries with ddcutil?

FYI, i2c-tools is available in LE.

rockowitz, 2025-04-19:
I have not tried to package shared libraries with ddcutil.  It would require non static versions of the libraries you want to package, and modifying the Makefile.am files.  

I'd suggest building your own copy of libjansson from source.

jansson is used in several parts of ddcutil.   Making it possible to build without those parts (i.e. with a --disable-jansson configure option) is a significant task, which is hard to justify  for LE.  There are many changes on the enhancement/todo list with greater impact.   Sorry.

ChaosNicro, 2025-04-24:
That is completely reasonable, I'll see about packing libjannson into an LE container. I didn't mean for you to alter ddutil for use with LE, I just thought it might be as simple as a parameter change on my side. No worries.

closed 2025-04-25 by rockowitz