rockowitz/ddcutil #511: Bug in display_lock.c (potentially infinite loop)

stefan-wltr, 2025-03-17:
https://github.com/rockowitz/ddcutil/blob/b15cc58c0b78d473b7f71e8bf08566293143e24b/src/base/display_lock.c#L230-L241
`total_wait_millisec` is initialized to 0, but never changed afterwards, whence the following while loop will run forever if no lock can be acquired.

(I came across this while debugging an issue with KDE powerdevil, whose main thread was hanging in this loop.)

rockowitz, 2025-03-18:
Fixed in 2.2.1-dev

stefan-wltr, 2025-03-18:
That code sample is from 2.2.1-dev. Maybe you fixed some related issue, but that while condition is still broken.

rockowitz, 2025-03-18:
Are you saying that commit bf6c55 did not fix the bug?

stefan-wltr, 2025-03-18:
I can't find that commit. The last commit on 2.2.1-dev is b15cc58 from Mar 12.

rockowitz, 2025-03-18:
Should be there now.

closed 2025-03-18 by stefan-wltr