rockowitz/ddcutil #514: setting brightness works reliably with skip-ddc-checks on BenQ GW2760

madroach, 2025-04-05:
I'm just reporting this in case in helps tracking down other issues or making ddcutil more resilient.

I can set brightness reliably with
`ddcutil -b 9 --skip-ddc-checks setvcp 10 0`
But without `--skip-ddc-checks` it fails reliably:
```
% ddcutil -b 9 --ddcdata setvcp 10 0
DDC: Invalid data length in packet: 111 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 6f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Unexpected source address 0x02, should be 0x6e
DDC: i2c_response_bytes: 02 00 0f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Invalid data length in packet: 104 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 68 82 00 10 00 00 64 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: Invalid data length in packet: 111 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 6f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Invalid data length in packet: 104 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 68 82 00 10 00 00 64 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC: Invalid data length in packet: 104 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 68 82 00 10 00 00 64 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC: Invalid data length in packet: 111 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 6f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
DDC: Invalid data length in packet: 104 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 68 82 00 10 00 00 64 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC: Invalid data length in packet: 104 exceeds MAX_DDC_DATA_SIZE
DDC: i2c_response_bytes: 6e 68 82 00 10 00 00 64 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DDC communication failed for monitor on bus /dev/i2c-9
```

Here's what probe does:
```
% ddcutil -b 9 --skip-ddc-checks probe
VCP (aka MCCS) version for display is undetected or less than 2.0. Interpretation may not be accurate.

EDID version: 1.3
Mfg id: BNQ, model: BenQ GW2760, sn: 55F05685019
Product code: 30918, binary serial number 21573 (0x00005445)

How display reports unsupported feature: DREF_DDC_USES_DDC_FLAG_FOR_UNSUPPORTED

Capabilities for display on bus /dev/i2c-9
Unable to get capabilities for monitor on Display_Handle[i2c-9: fd=3].  Maximum DDC retries exceeded.

Scanning all VCP feature codes for display Display_Handle[i2c-9: fd=3]
VCP code 0x02 (New control value             ): Maximum retries exceeded
VCP code 0x03 (Soft controls                 ): Maximum retries exceeded
VCP code 0x07 (Unknown feature               ): Maximum retries exceeded
VCP code 0x09 (Unknown feature               ): Maximum retries exceeded
VCP code 0x0b (Color temperature increment   ): Maximum retries exceeded
VCP code 0x0c (Color temperature request     ): Maximum retries exceeded
VCP code 0x0d (Unknown feature               ): Maximum retries exceeded
VCP code 0x0e (Clock                         ): Maximum retries exceeded
VCP code 0x0f (Unknown feature               ): Maximum retries exceeded
VCP code 0x10 (Brightness                    ): Maximum retries exceeded
VCP code 0x11 (Flesh tone enhancement        ): Maximum retries exceeded
^C
```

This is the monitor detected:
```
      I2C bus:  /dev/i2c-9
         DRM connector:                         card0-DP-4
         /sys/class/drm/card0-DP-4/dpms         On
         /sys/class/drm/card0-DP-4/enabled      enabled
         /sys/class/drm/card0-DP-4/status       connected
         /sys/class/drm/card0-DP-4/connector_id 120
         Driver:                                amdgpu
         EDID exists:                           true
         I2C address 0x37 (DDC)  responsive:    true
         Is LVDS or EDP display:                false
         Is laptop display by EDID:             false
         Is laptop display:                     false
         /sys/bus/i2c/devices/i2c-9/name        DPMST
         PCI device path:                       /sys/devices/pci0000:00/0000:00:08.1/0000:03:00.0/i2c-9
      EDID synopsis:
         Mfg id:               BNQ - UNK
         Model:                BenQ GW2760
         Product code:         30918  (0x78c6)
         Serial number:        55F05685019
         Binary serial number: 21573 (0x00005445)
         Manufacture year:     2015,  Week: 19
         EDID version:         1.3
         Extra descriptor:
         Video input definition:    0x80 - Digital Input
         Supported features:
            DPMS active-off
            Digital display type: RGB 4:4:4 + YCrCb 4:4:4
            Standard sRGB color space: False
         White x,y:        0.313, 0.329
         Red   x,y:        0.645, 0.334
         Green x,y:        0.331, 0.629
         Blue  x,y:        0.155, 0.047
         Extension blocks: 1
      EDID source: I2C
      EDID hex dump:
        +0          +4          +8          +c            0   4   8   c
+0000   00 ff ff ff ff ff ff 00 09 d1 c6 78 45 54 00 00   ...........xET..
+0010   13 19 01 03 80 3c 22 78 2e 2c c5 a5 55 54 a1 27   .....<"x.,..UT.'
+0020   0c 50 54 a5 6b 80 81 80 81 c0 81 00 a9 c0 b3 00   .PT.k...........
+0030   d1 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
+0040   45 00 56 50 21 00 00 1e 00 00 00 ff 00 35 35 46   E.VP!........55F
+0050   30 35 36 38 35 30 31 39 0a 20 00 00 00 fd 00 32   05685019. .....2
+0060   4c 1e 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   L.S...      ....
+0070   00 42 65 6e 51 20 47 57 32 37 36 30 0a 20 01 f2   .BenQ GW2760. ..
      VCP version:         Detection failed
      Controller mfg:      DDC communication failed
      Firmware version:    DDC communication failed
      Monitor uses invalid feature flag in DDC reply packet to indicate unsupported feature.
      Monitor Model Id:  BNQ-BenQ_GW2760-30918
      Feature definition file BNQ-BenQ_GW2760-30918.mccs not found.
```

rockowitz, 2025-04-13:
What the ***--ddcdata*** output shows is that the internal getvcp options performed during initialization are failing with data errors.  Also all the getvcp operations peformed by **probe** are failing.  So it's surprising is that you're not getting a verification error on **setvcp**.  (Might ***--noverify*** be specified in your ddcutilrc file?)

Have you tried option ***--sleep-multiplier 2.0***?

That said, it appears you're using a docking station, and i2c passthrough on docking stations can be problematic.  What is your hardware configuration?  Please run **sudo ddcutil interrogate** and submit the output as an attachment.

madroach, 2025-04-17:
I don't have a ddcutilrc. `--sleep-multiplier 2.0` makes no difference.

Yes, I use a docking station. The docking station is connected via USB-C to the laptop. The monitor is connected via HDMI to the docking station.

I just tried connecting the monitor directly to the laptop via HDMI. This shows the very same symptoms.

What I did notice right now is that about 1 in 20 attempts to set the brightness using `--skip-ddc-checks` has no effect, but no failure message. This is the same no matter whether the monitor is connected via docking station or directly connected.

Somehow I cannot upload an attachments to github. I put them here:
  * monitor connected to docking station: `ftp://gmerlin.de/pub/ddcutil.log.gz`
  * monitor connected directly to laptop: `ftp://gmerlin.de/pub/ddcutil-direct.log.gz`

rockowitz, 2025-04-19:
Unable to access the files.  I can connect to gmerlin.de, but access denied

madroach, 2025-04-19:
Ok. I also put them on http:

  * monitor connected to docking station: https://gmerlin.de/pub/ddcutil.log.gz
  * monitor connected directly to laptop: https://gmerlin.de/pub/ddcutil-direct.log.gz

rockowitz, 2025-04-28:
I'm afraid I don't have more to add.  For some reason you're getting DDC data errors.

madroach, 2025-04-28:
ok. I don't need this fixed.

closed 2025-04-28 by madroach