rockowitz/ddcutil #515: brightness changes failing (Asus VT229H)

briodan, 2025-04-12:
running into issues with setting brightness

```
ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
```

does not seem to work at all so mostly wanted to see if there is anything i can do about it or if its an issue with something not being supported by my monitor

[capabilities.log](https://github.com/user-attachments/files/19715573/capabilities.log)
[interrogate.log](https://github.com/user-attachments/files/19715572/interrogate.log)
[probe.log](https://github.com/user-attachments/files/19715574/probe.log)

rockowitz, 2025-04-13:
When you issue the **setvcp** command, is it the case that the screen brightness changes but verification fails, or that the screen brightness does not change?  There are monitors which change brightness but do not return the new value for **getvcp**.  Conversely there are monitors that do not change brightness but that report the new value.

If ti's the case that **setvcp** changes the screen brightness, does **getvcp** report the new value?

If it's simply the case that the monitor changes brightness but verification fails, consider using option ***--noverify***. 

Having said all that, I see you're running on a Raspberry Pi using driver vc4_v3d.  Your logs are full of "vc4-drm gpu: swiotlb buffer is full" messages.  So I suspect a driver issue.

briodan, 2025-04-13:
So yeah the brightness does not change when issuing the command.

It is a raspberry pi 4 using the latest updates on Debian bookworm, is there a specific version of the driver I need? I looked at the pi 4 instructions but those seem like they are for Debian 11 bullseye as I don’t have the config options in there.

briodan, 2025-04-14:
did a firmware update earlier today and got some mixed results

i was able to get a few commands to return "Verification succeeded" but also a bunch of verification failed. 
Looks to be working on setting the brightness when verification passes, however it does fail sometimes and when it fails brightness does not change.
Not sure if maybe its because i maybe run the commands too quickly after each other.

```
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 100
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 99
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Verification succeeded
pi@touch:~ $ ddcutil --verbose -d 1 getvcp 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs

Getting data for non-table VCP code 0x10 - Brightness:
Raw value: opcode=0x10, mh=0x00, ml=0x64, sh=0x00, sl=0x63, max_val=100 (0x0064), cur_val=99 (0x0063)
VCP code 0x10 (Brightness                    ): current value =    99, max value =   100
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 20
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 getvcp 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs

Getting data for non-table VCP code 0x10 - Brightness:
Raw value: opcode=0x10, mh=0x00, ml=0x64, sh=0x00, sl=0x63, max_val=100 (0x0064), cur_val=99 (0x0063)
VCP code 0x10 (Brightness                    ): current value =    99, max value =   100
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Verification succeeded
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 50
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Verification succeeded
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 99
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Verification succeeded
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Current value does not match value set.
Verification failed for feature 10
pi@touch:~ $ ddcutil --verbose -d 1 setvcp 10 10
Feature definition file not found: AUS-ASUS_VT229-8897.mccs
Verifying that value of feature 0x10 successfully set...
Verification succeeded
```

briodan, 2025-04-15:
seems to be working fine, as long as i dont send two commands in less then 30 seconds so i'm closing this out

closed 2025-04-15 by briodan