rockowitz/ddcutil #516: Brightness control issue with SAM-Odyssey_G5-29862 (Monitor Model ID)

JohanChane, 2025-04-16:
After booting, I cannot adjust the brightness. However, after setting the brightness using the monitor's built-in controls first, then I'm able to adjust the brightness. It's weird.

[detect.log](https://github.com/user-attachments/files/19771721/detect.log)

[capabilities.log](https://github.com/user-attachments/files/19771739/capabilities.log)

[version.log](https://github.com/user-attachments/files/19771744/version.log)

[environment.log](https://github.com/user-attachments/files/19771746/environment.log)

[interrogate.log](https://github.com/user-attachments/files/19771748/interrogate.log)

[probe.log](https://github.com/user-attachments/files/19771752/probe.log)

rockowitz, 2025-04-17:
The Samsung Odyssey seems to be a problematic monitor.  See for example issue #396 .  Most likely you start in a mode that disables brightness changes.  Take a look at feature xDC  (Display Mode).  Though it's not defined in the capabilities string, it does appear to be responsive  Changing its value may allow you to change the brightness.   Also, look at feature x14 (Color Preset).   You may need to set it to x0B (User 1) or x0C (User 2).

JohanChane, 2025-04-17:
I changed the picture mode to custom.  

When I modified the picture mode, x14 only had two cases:  

```sh  
~ ❯ ddcutil getvcp x14    
VCP code 0x14 (Select color preset           ): Display Native (sl=0x02)  
~ ❯ ddcutil getvcp x14  
VCP code 0x14 (Select color preset           ): sRGB (sl=0x01)  
```  

I couldn't change x14 to x0B (User 1) or x0C (User 2) via the `ddcutil` command.

JohanChane, 2025-04-17:
I've discovered that after each monitor power-off, I can regain ddcutil brightness control simply by opening and closing the monitor's OSD the `menu` option (without making any actual changes) using its physical controls.

JohanChane, 2025-04-18:
After updating the monitor's firmware from the official website, the issue still persists.
Model: LS27DG502ECXXF
https://www.samsung.com.cn/support/user-manuals-and-guide/