rockowitz/ddcutil #517: [amd] Acer G226HQL brightness slider in kde not detected anymore

renner0e, 2025-04-16:
Bazzite, Linux 6.13.9
It used to work a couple weeks ago for this monitor
My hardware setup didn't change.

```
❯ rpm -qi ddcutil-2.1.4-2.fc42.x86_64
Name        : ddcutil
Version     : 2.1.4
Release     : 2.fc42
Architecture: x86_64
Install Date: Mo 14 Apr 2025 04:18:16 CEST
Group       : Unspecified
Size        : 1174544
License     : GPL-2.0-or-later
Signature   : RSA/SHA256, Do 16 Jan 2025 17:22:24 CET, Key ID c8ac4916105ef944
Source RPM  : ddcutil-2.1.4-2.fc42.src.rpm
Build Date  : Do 16 Jan 2025 16:33:48 CET
Build Host  : buildvm-x86-29.iad2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : http://www.ddcutil.com
Bug URL     : https://bugz.fedoraproject.org/ddcutil
Summary     : Query and update monitor settings

```

[detect.txt](https://github.com/user-attachments/files/19785400/detect.txt)
[interrogate.txt](https://github.com/user-attachments/files/19785419/interrogate.txt)

rockowitz, 2025-04-17:
What slider are you referring to?

I see that you're on the newly released Fedora 42.  Does the change coincide with moving to the new Fedora release?

renner0e, 2025-04-17:
> What slider are you referring to?

oh yeah, this one here in kde for changing the brightness of the display

[image]

> I see that you're on the newly released Fedora 42. Does the change coincide with moving to the new Fedora release?

nope sadly not, will do a little bit more troubleshooting, maybe it's a newer kernel that broke it? (6.13.8 -> 6.13.9)
What is usually the cause for this just disappearing?

rockowitz, 2025-04-19:
I'm seeing the same thing on Fedora 42, KDE Plasma 6.3.80 (built from repos).   Highly unlikely that this is kernel related.   Fedora 42 upgraded Plasma, and also ddcutil.  Will investigate further.

renner0e, 2025-04-27:
My investigation so far tested with live ISOs:
KDE Neon unstable works, so not a Plasma bug
Fedora KDE 42: also works not a Fedora bug
Universal Blue's Aurora (based on F42): also works

renner0e, 2025-04-27:
it's probably a kde bug

rockowitz, 2025-04-28:
As I noted, I was seeing the same problem.  Then the sliders appeared again and I'm not sure what changed.

On my Fedora 42 system, I can log into a Plasma-Wayland session (using the Fedora provided Plasma), and also into development Plasma-Wayland and Plasma-X11 sessions that I've built.  As for libddcutil, I can use either the one in Fedora 42 (incorrectly reported as ddcutil 2.2.0-dev - it's actually 2.2.0) or my local copy built from ddcutil 2.2.1-dev. All of them work.

If you haven't already, I suggest you try "journalctl -b -t org_kde_powerdevil".  It may give a clue as to what is going on.

renner0e, 2025-04-28:
> If you haven't already, I suggest you try "journalctl -b -t org_kde_powerdevil". It may give a clue as to what is going on.

I completely forgot about powerdevil lol
```
Apr 27 19:11:49 org_kde_powerdevil[6105]: Initializing libddcutil.  ddcutil version: 2.1.4, shared library: /usr/lib64/libddcutil.so.5.1.2
Apr 27 19:11:49 org_kde_powerdevil[6105]: Options passed from client:
Apr 27 19:11:49 org_kde_powerdevil[6105]: Applying combined options:
Apr 27 19:11:51 org_kde_powerdevil[6105]: busno=7, sleep-multiplier =  2,00. Testing for supported feature 0x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_DDC_DATA, DDCRC_READ_ALL_ZERO(9)]
Apr 27 19:11:51 org_kde_powerdevil[6105]: busno=9, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:11:51 org_kde_powerdevil[6105]: busno=9, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:11:51 org_kde_powerdevil[6105]: busno=9, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:11:51 org_kde_powerdevil[6105]: busno=9, All features that should not exist detected. Monitor does not indicate unsupported
Apr 27 19:11:51 org_kde_powerdevil[6105]: Library initialization complete.
Apr 27 19:11:51 org_kde_powerdevil[6105]: org.kde.powerdevil: [DDCutilDetector]: Failed to initialize callback
Apr 27 19:11:51 org_kde_powerdevil[6105]: org.kde.powerdevil: org.kde.powerdevil.chargethresholdhelper.getthreshold failed "Charge thresholds are not supported by the kernel for this hardware"
Apr 27 19:11:51 org_kde_powerdevil[6105]: org.kde.powerdevil: org.kde.powerdevil.backlighthelper.brightness failed
Apr 27 19:11:52 org_kde_powerdevil[6105]: org.kde.powerdevil: Handle button events action could not check for screen configuration
Apr 27 19:11:52 org_kde_powerdevil[6105]: org.kde.powerdevil: org.kde.powerdevil.chargethresholdhelper.getthreshold failed "Charge thresholds are not supported by the kernel for this hardware"
Apr 27 19:12:15 org_kde_powerdevil[7896]: Initializing libddcutil.  ddcutil version: 2.1.4, shared library: /usr/lib64/libddcutil.so.5.1.2
Apr 27 19:12:15 org_kde_powerdevil[7896]: Options passed from client:
Apr 27 19:12:15 org_kde_powerdevil[7896]: Applying combined options:
Apr 27 19:12:16 org_kde_powerdevil[7896]: busno=7, sleep-multiplier =  2.00. Testing for supported feature 0x10 returned Error_Info[DDCRC_RETRIES in ddc_write_read_with_retry, causes: DDCRC_READ_ALL_ZERO(2), DDCRC_DDC_DATA, DDCRC_READ_ALL_ZERO(7)]
Apr 27 19:12:17 org_kde_powerdevil[7896]: busno=9, Feature 0xdd should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:12:17 org_kde_powerdevil[7896]: busno=9, Feature 0x41 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:12:17 org_kde_powerdevil[7896]: busno=9, Feature 0x00 should not exist but ddc_get_nontable_vcp_value() succeeds, returning mh=0xff ml=0xff sh=0x00 sl=0xff
Apr 27 19:12:17 org_kde_powerdevil[7896]: busno=9, All features that should not exist detected. Monitor does not indicate unsupported
Apr 27 19:12:17 org_kde_powerdevil[7896]: Library initialization complete.
Apr 27 19:12:17 org_kde_powerdevil[7896]: org.kde.powerdevil: [DDCutilDetector]: Failed to initialize callback
Apr 27 19:12:17 org_kde_powerdevil[7896]: org.kde.powerdevil: org.kde.powerdevil.chargethresholdhelper.getthreshold failed "Charge thresholds are not supported by the kernel for this hardware"
Apr 27 19:12:17 org_kde_powerdevil[7896]: org.kde.powerdevil: org.kde.powerdevil.backlighthelper.brightness failed
Apr 27 19:12:17 org_kde_powerdevil[7896]: org.kde.powerdevil: Handle button events action could not check for screen configuration
Apr 27 19:12:17 org_kde_powerdevil[7896]: org.kde.powerdevil: org.kde.powerdevil.chargethresholdhelper.getthreshold failed "Charge thresholds are not supported by the kernel for this hardware"
Apr 27 19:20:19 org_kde_powerdevil[6105]: The Wayland connection broke. Did the Wayland compositor die?
Apr 27 19:20:19 org_kde_powerdevil[6105]: qt.qpa.wayland: Attempting wayland reconnect
Apr 27 19:20:19 org_kde_powerdevil[6105]: Failed to write to the pipe: Ungültiger Dateideskriptor.
Apr 27 19:20:19 org_kde_powerdevil[6105]: qt.qpa.wayland: There are no outputs - creating placeholder screen
Apr 27 19:20:50 org_kde_powerdevil[6105]: The Wayland connection broke. Did the Wayland compositor die?
Apr 27 19:20:50 org_kde_powerdevil[6105]: qt.qpa.wayland: Attempting wayland reconnect
Apr 27 19:20:50 org_kde_powerdevil[6105]: Failed to write to the pipe: Ungültiger Dateideskriptor.
Apr 27 19:20:50 org_kde_powerdevil[6105]: qt.qpa.wayland: There are no outputs - creating placeholder screen
Apr 27 19:24:53 org_kde_powerdevil[6105]: The Wayland connection broke. Did the Wayland compositor die?
Apr 27 19:24:53 org_kde_powerdevil[6105]: qt.qpa.wayland: Attempting wayland reconnect
Apr 27 19:24:53 org_kde_powerdevil[6105]: Failed to write to the pipe: Ungültiger Dateideskriptor.
Apr 27 19:24:53 org_kde_powerdevil[6105]: qt.qpa.wayland: There are no outputs - creating placeholder screen
Apr 27 19:25:38 org_kde_powerdevil[6105]: The Wayland connection broke. Did the Wayland compositor die?
Apr 27 19:25:38 org_kde_powerdevil[6105]: qt.qpa.wayland: Attempting wayland reconnect
[... cut ...]
Apr 27 19:29:59 org_kde_powerdevil[19918]: org.kde.powerdevil: [DDCutilDisplay]: ddca_get_non_table_vcp_value -3007
Apr 27 19:29:59 org_kde_powerdevil[19918]: org.kde.powerdevil: [DDCutilDisplay]: "Acer G226HQL" failed to initialize DDC/CI brightness
Apr 27 19:29:59 org_kde_powerdevil[19918]: dh=Display_Handle[i2c-9: fd=28], Replacing adjusted sleep multiplier 0.20 with 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
Apr 27 19:30:00 org_kde_powerdevil[19918]: dh=Display_Handle[i2c-9: fd=28], Replacing adjusted sleep multiplier 0.10 with 1.00 for SE_POST_WRITE or SE_POST_SAVE_SETTINGS
```

renner0e, 2025-04-29:
logs from the new user where it works:

[interrogate.txt](https://github.com/user-attachments/files/19968032/428a00d4.txt)

[powerdevil_logs.txt](https://github.com/user-attachments/files/19968035/aa07cf3c.txt)

rockowitz, 2025-05-05:
Your latest logs show ddcutil failing.  Given what I'm seeing, irrespective of any powerdevil problems, command line ddcutil should be failing as well.  The reply packets on /dev/i2c-7 are consistently all 0, indicating a DDC communication error. 

If I'm interpreting the interrogate output correctly, there is an XFX video card in the system, with one monitor connected to the HDMI connector on the motherboard, the other connected to the video card.  You might try connecting both monitors to the video card.

renner0e, 2025-05-05:
> If I'm interpreting the interrogate output correctly, there is an XFX video card in the system, with one monitor connected to the HDMI connector on the motherboard, the other connected to the video card. You might try connecting both monitors to the video card.

It's an RX 6750XT from ASRock.
I do not use my iGPU at all. It's disabled in the bios.
Both monitors are connected to the proper Graphics card.