rockowitz/ddcutil #518: Expose drm connector in API

Zamundaaa, 2025-04-16:
Currently in Plasma we attempt to figure out which display ref belongs to which output by matching the EDID, but when there's two displays with the same EDID, that ends up basically random.

If you could expose the drm connector path, we could always assign the drm devices to the correct output (besides race conditions with hotplugs, but it's still better than totally random).

rockowitz, 2025-04-18:
It's certainly possible to expose the drm connector path (though not trivial to do so in a way that makes the API change backwardly compatible).  However, it's only a partial solution to your problem.  Depending on the video driver and driver version, /sys may or may not be populated with the requisite attributes to definitively associate a /dev/i2c device (and hence the DisplayRef) with a DRM connector.  If the attributes exist, ddcutil uses them to make the connection;  if not, it falls back to EDID matching.  

For recent amdgpu and i915 drivers, the needed attibutes exist.  For the Nvidia proprietary driver, they do not.

rockowitz, 2025-04-26:
Struct DDCA_Display_Info2 extends DDCA_Display_Info, adding members drm_card_connector, drm_card_connector_found_by, and drm_connector_id.  Functions ddca_get_display_info2(), ddca_free_display_info2() and ddca_report_display_info2() have been added for using DDCA_Display_Info2.  The corresponding DDCA_Display_Info functions continue to exist, but their use iis deprecated.  

Note that there is no DDCA_Display_Info2 variant for  ddca_get_display_info_list2() (which returns a list of  DDCA_Display_Info) and ddca_report_display_info_list(), as  ddca_get_display_refs()/ddca_get_display_info2() is the recmmended way to retrieve display reference information.

Zamundaaa, 2025-04-29:
Cool, thank you!

That this information isn't 100% reliable is ok, something is better than nothing :)

closed 2025-04-29 by Zamundaaa