rockowitz/ddcutil #540: F42 on a Framework 16 returns "display not found" for `ddcutil capabilities`, even with `sudo` prepended.

RokeJulianLockhart, 2025-08-14:
#### An Example

<blockquote>

~~~txt
RokeJulianLockhart@Beedell:~$ sudo ddcutil capabilities
Display not found
~~~

</blockquote>

#### My Environment

~~~sh
#!/usr/bin/env bash
hwinfo --monitor
~~~

<blockquote>

~~~YAML
46: None 00.0: 10002 LCD Monitor                                
  [Created at monitor.125]
  Unique ID: rdCR.HLVCnEUYHWA
  Parent ID: 0Gk3.6FldxGZRsU8
  Hardware Class: monitor
  Model: "BOE CQ LCD Monitor"
  Vendor: BOE "BOE CQ"
  Device: eisa 0x0bc9 
  Serial ID: "0"
  Resolution: 2560x1600@60Hz
  Size: 345x215 mm
  Year of Manufacture: 2022
  Week of Manufacture: 48
  Detailed Timings #0:
     Resolution: 2560x1600
     Horizontal: 2560 2608 2640 2720 (+48 +80 +160) -hsync
       Vertical: 1600 1663 1669 1760 (+63 +69 +160) +vsync
    Frequencies: 287.24 MHz, 105.60 kHz, 60.00 Hz
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #21 (VGA compatible controller)
~~~

</blockquote>

~~~sh
#!/usr/bin/env bash
dnf5 info $(rpm -qa ddcutil)
~~~

<blockquote>

~~~YAML
Name            : ddcutil
Epoch           : 0
Version         : 2.1.4
Release         : 2.fc42
Architecture    : x86_64
Installed size  : 1.1 MiB
Source          : ddcutil-2.1.4-2.fc42.src.rpm
From repository : <unknown>
URL             : http://www.ddcutil.com
Vendor          : Fedora Project

~~~

</blockquote>

RokeJulianLockhart, 2025-08-15:
@rockowitz, do you believe that this message can be improved? Considering that KDE Plasma somehow manages to modify the display's brightness, I was under the impression that `ddcutil` was how it (presumably, PowerDevil) accomplished this.

rockowitz, 2025-08-15:
Laptop display brightness is controlled using ACPI, not DDC/CI.  Powerdevil uses ACPI, not ddcutil, for laptop displays.

Following the practice of Argyll CMS, displays supporting DDC/CI are numbered from 1.  The display number is reported by the **detect** command.  If no explicit display selection option (e.g. ***--display***, ***--bus***, ***--model***, ***--sn***, ***--edid***) is specified on a command that operates on a valid display, option ***--display 1*** is assumed.  But there is no display found that matches that selection criterion. Hence the message "Display not found".

closed 2026-01-19 by rockowitz