rockowitz/ddcutil #546: running environment cmd drops a display from detection

laur89, 2025-09-02:
Hi again,

### Issue

upon executing `ddcutil environment --very-verbose`, _detect_ no longer shows the single valid display that's connected.
Only way to get it detected again is to physically reconnect the monitor.

### Repro (seems to reproduce consistently):

1.
```
$ ddcutil --brief detect
ddcutil --brief detect
Invalid display
   I2C bus:          /dev/i2c-4
   DRM connector:    card0-eDP-1
   drm_connector_id: 100
   Monitor:          LEN::

Invalid display
   I2C bus:          /dev/i2c-8
   drm_connector_id: 0
   Monitor:          AUO::

Display 1                        <--  this is the display I'm talking about
   I2C bus:          /dev/i2c-11
   DRM connector:    card0-DP-6
   drm_connector_id: 0
   Monitor:          AUS:PA278QV:L8LMQS127231
```

2. run `ddcutil detect --very-verbose --f2` (output in [gist](https://gist.github.com/laur89/641e240d30fd1cc3f7944d158e7cd441#file-ddcutil-detect-verbose-f2-first))
3. run `ddcutil environment --very-verbose` (output in [gist](https://gist.github.com/laur89/641e240d30fd1cc3f7944d158e7cd441#file-ddcutil-env-verbose))
    - this is the command that causes `Display 1` to be lost.
4. run `ddcutil detect --very-verbose --f2` (output in [gist](https://gist.github.com/laur89/641e240d30fd1cc3f7944d158e7cd441#file-ddcutil-detect-verbose-f2-second))
5. 
```
$ ddcutil --brief detect
Invalid display
   I2C bus:          /dev/i2c-4
   DRM connector:    card0-eDP-1
   drm_connector_id: 100
   Monitor:          LEN::

Invalid display
   I2C bus:          /dev/i2c-8
   drm_connector_id: 0
   Monitor:          AUO::
```

Note `Display 1` is now droped.

---

ThinkPad [p14s](https://psref.lenovo.com/syspool/Sys/PDF/ThinkPad/ThinkPad_P14s_Gen_1_AMD/ThinkPad_P14s_Gen_1_AMD_Spec.PDF)  docked to [40AJ](https://support.lenovo.com/es/en/solutions/pd500173-thinkpad-ultra-docking-station-overview-and-service-parts) docking station.
Running debian testing, ddcutil `2.2.1` installed from debian repos.

rockowitz, 2025-09-02:
Do you encounter the problem without option ***--very-verbose***?

rockowitz, 2025-09-02:
Disregard the question about ***--very-verbose***.  Irrelevant.  Do you encounter the problem if the monitor is plugged directly into the laptop when the docking station is/is not also connected?

laur89, 2025-09-03:
Sorry I had a brainfart earlier - meant to say it works _correctly_ when hooked to laptop directly.

laur89, 2025-09-04:
> The HDMI port is not the only video port on the laptop

True, but don't have a cable for that.

But yes, sounds like similar issue I reported handful of years ago. Docking stations be docking. If keeping this issue open is not beneficial, please close it.